prusa3d/PrusaSlicer #10058: [REQUEST] Skirt extrusion width

Merlin246, 2023-03-14:
**Is your feature request related to a problem? Please describe.**
I use the skirt typically to prime my nozzle, I would like to extrude thicker lines for my skirt but cannot as there is no distinct settings for skirt line width.

**Describe the solution you'd like**
Under Print Settings > Advanced > Extrusion width; add an additional field for skirt line width.

**Describe how it would work**
Uses the prescribed line width for skirts, if 0 uses the first layer line width.

**Describe alternatives you've considered**
Using SuperSlicer.

kubispe1, 2023-03-15:
Thanks for your request. 
For priming purposes, you can also use two loops of skirt???...

Merlin246, 2023-03-15:
@kubispe1 you certainly could, however if the skirt is more than 1 layer high it's just a waste of plastic, which may be somewhat expensive if you're using something like a carbon-fibre filled filament. A solution to this would be having a configurable number of loops or minimal filament extrusion length for the first layer. SuperSlicer has a "skirt brim" implemented in their expert mode under Print Settings > Skirt & Brim > Skirt > Brim. I've used it in the past, its a nice feature to have and hopefully wouldn't be too difficult to integrate into PrusaSlicer it's based off PrusaSlicer.

I also find that thicker lines are better for priming and cleaning the nozzle, similar to the default start-gcode of the MK3S. Perhaps a separate feature that adds something similar to the start of the print (whether completely separate or it leads into the skirt) would be useful.

Merlin246, 2023-03-15:
@neophyl I understand and use those settings already. I am asking for those additional settings. The settings I was talking about in my previous reply would be separate settings for only the first layer. ie having 3 loops on first layer but only 1 loop on all subsequent layers. Hope this reply clarifies.

neophyl, 2023-03-15:
Still not sure of the issue, the Loops minimum is applied to the first layer, its a separate setting to height,  you just posted above - "A solution to this would be having a configurable number of loops or minimal filament extrusion length for the first layer."  so the settings already do what you describe as a solution ?  
Just set 1 loop high and the minimal filament length and it will prime just fine and not waste material which seems to be your prime concern.
For those who's primary concern is skirt removal being difficult at at only 1 layer high they can set the height value too and so everyone should be able to get what they need already.

Merlin246, 2023-03-15:
@neophyl I cannot set the slicer to generate a 3-loop skirt on the first layer and 1-loop skirt for subsequent layers. This is what I'm aiming for. I do not want the additional loops as you suggested on layers 2+.

closed 2026-09-29 by prusaslicer-bot[bot]