prusa3d/PrusaSlicer #10061: SLicer will not connect to Duet SBC with password

stuartofmt, 2023-03-14:
### Description of the bug

I am running Duet 3Mini5+ in SBC (Single Board Computer) mode.
Pruse Slicer is unable to connect  if a password is set on the Duet.  It will connect and is able to transfer files if there is no password.

In SBC mode, when a password is used, the Duet requires a session key in the http header. 

Below, for reference, is skeletal python code that illustrates a working connection .  Subsequent calls must include urlHeaders if a password is used.

`
global urlHeaders, timelimit, duetip
duetip = '192.168.30.30'
urlHeaders  = {}
timelimit = 5
duetip = '192/168.30.30'

password = 'mypassword'

def connect()
     url = ('http://' + duetip + '/machine/connect?password=' + password) 
     r = requests.get(url, timeout=timelimit, headers=urlHeaders)
     code = r.status_code
     if code == 200:
          # Connected to SBC printer
          j = json.loads(r.text)
          sessionKey = j['sessionKey']
         urlHeaders = {'X-Session-Key': sessionKey}
     # trap other error codes    

# Get connected and get session key
connect()
#  Follow on calls  .......
`

### Project file & How to reproduce

100% reproducuble by selecting Test 
[image]

On the Duet (DSF) side, the error is reported as

`
Mar 14 13:11:44 srsender DuetWebServer[645]: DuetWebServer.Authorization.SessionKeyAuthenticationHandler[7] SessionKey was not authenticated. Failure message: Missing X-Session-Key header
`

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha5+win64-202303061442_signed

### Operating system

Win11

### Printer model

Ender 5 Pro

Jan-Soustruznik, 2023-03-24:
Issue created SPE-1611

kocikdav, 2023-10-19:
Hello @stuartofmt
Would you test for us this [test build](https://storage.cloud.google.com/prusa3d-content-prod-14e8-slicerbuilds/PrusaSlicer-2.6.1%2B1-dk-duet-win64-g5664cc22e-202310181503.zip)? Unfortunately we do not a machine where we could test these changes. If you  confirm it works, we will include these changes into next release.
The changes can be seen in [commit](https://github.com/prusa3d/PrusaSlicer/commit/5664cc22e2fc5d1b04ecab663be5abfe6d4880ed).

kocikdav, 2023-10-23:
Oh yes, I forgot to URL encode the password. Please try [this build](https://slicerbuilds.prusa3d.com/PrusaSlicer-2.7.0-alpha1%2B2-dk-duet-win64-g67d366b0b-202310231157.zip).

kubispe1, 2023-10-26:
@stuartofmt please can you try ??

stuartofmt, 2023-10-26:
Looks to be working fine.  I tried a number of cominations, all with success.  Thanks!

kocikdav, 2023-10-31:
Thanks! Will be part of next release.

closed 2023-10-31 by kocikdav