prusa3d/PrusaSlicer #10063: Can't save sections from "CUT" performed

LemleyW, 2023-03-14:
### Description of the bug

What is the Problem?  Can't save the two pieces of the "CUT" function..
What did you expect?  I expected to save the two pieces so I could print them as individual STLs because they both don't fit on the build plate at the same time.

[image]

### Project file & How to reproduce

[bordeaux-the-octopus.zip](https://github.com/prusa3d/PrusaSlicer/files/10974671/bordeaux-the-octopus.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-Alpha4+win64 

### Operating system

Windows 10

### Printer model

Prusa i3 MK3S+ MMU2S Single

kubispe1, 2023-03-17:
Everything works for me only holes of connectors are not included (negative volumes are not possible to export).
In case try to describe the issue more

LemleyW, 2023-03-17:
Hi,   The Octopus Wine Bottle Holder  is 209.14 in "Z" axis close to the 210 for my Prusa printer.  I wanted to cut the Octopus just under the eye and save the two parts as separate STL's.  When I do the "CUT" I get two parts side by side which puts one of the parts off the print area.  I try and click on the part off the print bed and save it but I can't.  It's a good possibility I am just not following the correct procedure but I don't see instructions for how to do what I am trying. For smaller STL's where the CUT parts can be placed on the print bed at the same time it doesn't seem to be a problem.  It's just this big one.
Regards,Mike Wright

kubispe1, 2023-03-20:
here is my procedure. Check it and try, please. 
[image]

closed 2026-09-29 by prusaslicer-bot[bot]