prusa3d/PrusaSlicer #10065: After Nvidia driver update, all running PrusaSlicer instances silently crash when focused

talaman4eg, 2023-03-15:
### Description of the bug

Running PrusaSlicer instances sometimes crash when swithing focus on them. Before the crash they were running for a few days with no issues. It occasionally happened before, with different versions, but right now it happened right after Nvidia drivers update. It is possible, that previous occasions had the same reason. When I get the next driver update, I will verify additionally.

Unfortunately, no screenshot available due to the nature of the bug. 

### Project file & How to reproduce

Run PrusaSlicer (in my case, few instances for a few days, multiple system hybernates, no issues)
Update Nvidia drivers (Nvidia Geforce experience - Quick Update). Wait until update completes
Focus any PrusaSlicer instance - it silently dissappears from the taskbar, one by one, after focusing each instance

Currently I'm using 2.6.0-alpha5+win64, but it happened with 2.5 stable for sure, and probably with previous versions as well. I will verify when next driver update arrives.

I don't have project files because I didn't save them. There were sets of different STL's, I tested different filaments with different, but very similar settings. I replaced models, tweaked some settings on the fly and printed results, then repeated. I tried to reproduce one of the projects, but I might be missing some settings changes or models were different. There were 4 or 5 instances opened, with similar models/settings, and they all crashed one by one after video driver update.

Please, let me know if there any logs I can attach.

My System info:

PrusaSlicer
Version:   2.6.0-alpha5+win64
Build:     PrusaSlicer-2.6.0-alpha5+win64-202303061442

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19045), 64-bit edition
Total RAM size [MB]: 25,702MB
OpenGL installation
GL version:   4.6.0
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce GTX 1080/PCIe/SSE2
GLSL version: 4.60.0
Textures compression:       Enabled
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
[calicat_vertically_stackable.zip](https://github.com/prusa3d/PrusaSlicer/files/10976319/calicat_vertically_stackable.zip)

GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
[... 353 lines cut ...]
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha5+win64

### Operating system

Windows 10 (build 19045), 64-bit edition

### Printer model

Prusa i3 MK3s

talaman4eg, 2023-03-15:
Also, please let me know if I can do any additional actions that can help you to debug (logging, recording or any other) during the next drivers update. I am going to run instances of 2.5 and 2.6-alpha (I have them installed), then update drivers, then see if they crash.

Matrix0815-Mod, 2023-03-18:
hi ich habe mir auch die Alpha5 geladen sie stürzt bei mir nur auf dem AMD ab es schreibt jedes mal Shader fehler,
ichhabe auch die 2.5.1 drauf die läuft stabil. 
wenn ich den Benutzer unter Windows 10 Wechsel laufen beide Versionen Stabil auch die alpha5
ich find den Fehler nicht.
ich habe auf meinem 2. Pc beide installiert keine Probleme.
nur auf dem AMD siehe Log Datei
[log.txt](https://github.com/prusa3d/PrusaSlicer/files/11007926/log.txt)

ich habe auch versucht aus dem Intel PC die gesamten Ordner zu zippen und auf dem AMD entpackt die Alpha5 startet trotzdem nicht. , nur unter einem anderem Beutzerkonto.
ich habe auch Nvidia karten beim AMD RTX2070 S  , beim Intel PC GTX960 bei beiden ist Experance  v3.27.0112 und Treiber 531.29

talaman4eg, 2023-03-23:
Nvidia driver update just arrived and I verified the issue with both PrusaSlicer-2.6.0-alpha5+win64-202303061442 and PrusaSlicer-2.5.0+win64-202209060714
I opened 2 instances of each version: one empty and one with the project I attached. All of them quietly disappeared from the task bar during the driver update (NVIDIA GeForce Experience - Drivers - Express Installation)
Other running application were not affected.

closed 2026-09-29 by prusaslicer-bot[bot]