prusa3d/PrusaSlicer #10068: Inner brim is positioned wrong.

cmoiceanu, 2023-03-15:
### Description of the bug

I have an STL am trying to add inner brim to it, but the brim is placed away from desired object surface. See attached screen shot and the STL in question. 
I can also repro the problem with release slicer version 2.5.0+win64 but not in older Super Slicer Version 2.4.58.2

### Project file & How to reproduce

[invalid_inner_brim.zip](https://github.com/prusa3d/PrusaSlicer/files/10978369/invalid_inner_brim.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha5+win64

### Operating system

Windows 10 

### Printer model

 Voron 2.4 with afterburner

Pejfa, 2023-03-16:
Hello, @cmoiceanu, Why did you use for 0.4 nozzle the extrusion width 0.6 ?? If I change the extrusion with for first layer to 0.42 then problem disapears .. do you think that may be just with wrong settings for this model?

cmoiceanu, 2023-03-16:
Because it can be change - last I remember up to something like 240% of nozzle size - and mainly because I want to print faster.
 
However, that's beside the point.   Turning our attention to the STL sample I submitted. I have created a clean project, added the STL with **all default settings** for MK3s - the same problem persists - attaching new 3mf. 

This is a legit bug, and easy to reproduce in the slicer.  

I'm glad to know that changing the line width is a potential workaround (will have to try it myself),  
[invalid_inner_brim_with_default_prusa_profiles.zip](https://github.com/prusa3d/PrusaSlicer/files/10995805/invalid_inner_brim_with_default_prusa_profiles.zip)

closed 2026-09-29 by prusaslicer-bot[bot]