prusa3d/PrusaSlicer #10070: Turn off the stepper motors after loading the filament

Pitkox, 2023-03-15:
**Is your feature request related to a problem? Please describe.**
After loading new filament I need to shut down stepper motors from menu to manually pull up filament from extruder a little bit, to prevent licking when it's hot.

**Describe the solution you'd like**
Auto shut down motors after loading filament, or pull it a little bit up automatically after correct loading to prevent from licking out of the extruder.

neophyl, 2023-03-15:
Wrong repository, this is the slicer github.  Your request is firmware of the printer.  You don't mention what printer you have so pointing you to the correct place is a little hard.

Pitkox, 2023-03-15:
Oh my,.. youre right sorry, too many cards opened.
Printer is MK3S

closed 2023-03-15 by Pitkox