prusa3d/PrusaSlicer #10071: Auto sort doesn't account for brim

addohm, 2023-03-15:
### Description of the bug

When sorting, parts are placed too close to other parts that have brim\support and end up on top of the brim.

### Project file & How to reproduce

Nothing fancy about it.  With the auto sort spacing set to a low number (say 1 or 2mm), put several parts on the build plate and give all the parts a brim or support, then execute the auto sort and check the gcode preview.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha5

### Operating system

Windows 11

### Printer model

MK3S+

foreachthing, 2023-03-15:
> and end up on top of the brim

I do that on purpose, sometimes. Mostly with tiny parts. So, I like how it PS works.
You can manually change the offset in arrange (right click symbol)
[image]

jmd73, 2023-03-16:
As a fix, you can merge the all the objects after arranging them

closed 2026-09-29 by prusaslicer-bot[bot]