prusa3d/PrusaSlicer #10073: Thick bridges in bridge infill causes overextrusion

ddan1924, 2023-03-15:
### Description of the bug

It seems thick bridges always generate with 0.4 layer height. I slice my model with 0.14mm layer height for extra detail. So if the bridge infill generates, it will extrude for 0.4mm height when the actual layer height is only 0.14mm. Even worse, in this model there are regions where 2 or 3 thick bridges overlap, causing infill extruded for 1.2mm height in only 0.42 actual height. It also ignores the max layer height settings in the extruder tab, which is set to only 0.3mm. In the end it causes a lot of overextrusion.

Layer 1 of bridge infill:
[image]

Layer 2 of bridge infill:
[image]

Layer 3 of bridge infill:
[image]

### Project file & How to reproduce

Project File:
[2020treesgenericv2.zip](https://github.com/prusa3d/PrusaSlicer/files/10980495/2020treesgenericv2.zip)

Just Slice and set the legend in the preview to height

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha5

### Operating system

Windows 11

### Printer model

Sovol SV06 with Klipper

lukasmatena, 2023-04-02:
Thanks. Should be fixed in [2.6.0-alpha6](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha6). Closing. Let us know if the issue persists.

closed 2023-04-02 by lukasmatena

kcalbxof, 2023-10-24:
> Thanks. Should be fixed in [2.6.0-alpha6](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha6). Closing. Let us know if the issue persists.

Release 2.6.1+win64, issue appears to be not fixed. Moreover, i see that bridge_flow_ratio setting is responsible for adjusting layer height for bridges. Setting it to 0.7 made layer height 0.25mm as i had set in my profile.

Moreover, layer height constraint that is set in extruder tab is not working in this regard, still.