prusa3d/PrusaSlicer #10074: 2.6 alpha: bring back checkbox for ensure vertical shell thickness

meltiseugen, 2023-03-15:
### Description of the bug

Now vertical shell thickness is on by default and can not be turned off.

### Project file & How to reproduce

Have the checkbox to control the vertical shell thickness solid infill back

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6-alpha5

### Operating system

macOS

### Printer model

Anycubic Mega X

sarusani, 2023-03-16:
"The old "Ensure vertical shell thickness" parameter used to switch between the upstream Slic3r and Slic3r Prusa Edition algorithms to ensure vertical wall thickness. The new algorithm implemented in PrusaSlicer 2.6.0-alpha5 is significantly better than either of the two old algorithms; thus, the switch "Ensure vertical shell thickness" was removed."

eugenmeltis3pg, 2023-03-16:
> 

Yes, but what if I don't want to have those extra solid infills added in the slice?
Basically I just want an option to turn that on/off

meltiseugen, 2023-03-16:
> "The old "Ensure vertical shell thickness" parameter used to switch between the upstream Slic3r and Slic3r Prusa Edition algorithms to ensure vertical wall thickness. The new algorithm implemented [...]

Yes, but what if I don't want to have those extra solid infills added in the slice?
Basically I just want an option to turn that on/off

sarusani, 2023-03-16:
There was never an option to turn them off. The old box just switched between two different algorithms to calculate them. This is no longer needed since the new algorithm replaces both of them.

Snuff1eupagus, 2023-03-16:
@meltiseugen 

I agree with you, regardless of what ticking the radio button did in the past, the result\affect was it removed many areas of ensure vertical shell thickness. while there are workarounds that can help with this, it's proven very inconvenient to have to do compared to just being able to turn them off. There should be a button to remove them when unwanted. I would also say in this same paragraph that I'd like to have the control to switch back to the older bridging method, and seams from 2.4.2 when I need to. 

I don't understand why it has to be one way or the other, why can't we have both. Now we have "ensure vertical shell" that we have to jump through hoops to get rid of if we don't want/need it, Internal bridging that you can't control if/when we need to, and Seams that won't do anything we want them to. How are these improvements? when all your doing is taking the control from us and then saying to us "oh what did you need that for?"

meltiseugen, 2023-03-17:
I also agree, we need a way to disable this entirely.
I will change the issue title to reflect this accordingly.

sarusani, 2023-03-17:
Since it's currently behaving as it was intended by the developers, the correct way would be to close this bug ticket and create a feature request instead.

closed 2023-03-19 by meltiseugen

AudreyAP, 2023-06-29:
I can't seem to find a way to do this in 2.6 - not sure why this issue was closed. I cannot figure out how to get rid of this extra infill. 

Benchy example: 

[image]

Tupson444, 2023-06-30:
This issue is closed because another one is open: #10102