prusa3d/PrusaSlicer #10077: [2.6.0-alpha5] Nozzle hitting organic support

mfettig, 2023-03-15:
### Description of the bug

During printing there is a travel through a previously printed layer of an organic support. This broke the support off causing trouble with the print

[image]
[image]

On previous layers the supports were printed in a different order, avoiding the collision

### Project file & How to reproduce

[ProjectSettingsForBugReport.zip](https://github.com/prusa3d/PrusaSlicer/files/10984441/ProjectSettingsForBugReport.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha5+linux-x64-GTK3

### Operating system

Linux Mint 21

### Printer model

Anycubic Kobra Go

foreachthing, 2023-03-16:
Enable z-hop when printing orcanic support.
https://github.com/prusa3d/PrusaSlicer/issues/9826

ltlowe, 2023-03-16:
Enable z-hop is a work-around.  The slicer should ensure that it does not cross its own path on the same layer.

mfettig, 2023-03-16:
Thanks for the suggestion @foreachthing, enabling z-hop likely would help with getting this model printed

I have to agree with @ltlowe though, and hope that is the official stance as well. It would be disappointing to need z-hop enabled any time I want to use organic supports

I've printed 3 or 4 other models with organic supports without issue, but going forward it isn't reasonable to manually check every travel before printing to see if there is a collision after slicing, nor is it reasonable to fail part way through a print just to go back and enable z-hop. For me at least, that would make z-hop a requirement of printing with organic supports, which I hope is not going to be the case

onolox, 2023-03-26:
https://user-images.githubusercontent.com/11450681/227791993-1f83b7da-3ad5-44c8-872f-ded07505c1fb.mp4

onolox, 2023-03-28:
Like my video shows above, the engine to generate the travel of the PS is very dumb, I noted before that it does not generate an optimized path, it goes everywhere almost randomly, creating empty travels all around. Cura for example, has a very optimized travel sequence AND avoids hitting, by taking a detour or z-hop.
Common PS team, it's all open source.
Don't take mem wrong, I love PS but this is a game breaker.
I have an example of the problem of the PS, I had to improvise to put the support again in the print, PS is hitting the supports tall the time on this:

[image]

mfettig, 2023-06-06:
Using 2.6.0-beta4 I wasn't able to recreate this issue, so this appears to be resolved. Thank you!

closed 2023-06-06 by mfettig

xardael, 2024-02-23:
> This happens all the time with me. The engine don't take the organic supports into consideration. In overrall the travel engine is very dumb in prusaslicer. Time to go to cura... 

Cura v5.6.0 has the exact same bug, reported on github as well since May 2023.
https://github.com/Ultimaker/Cura/issues/15479

So I actually wanted to get PrusaSlicer instead and I ended up here... Seems like both using the same trave engine.
The organic supports are great idea but they are useless untill this will be resolved. Every 3rd print is ruined.

onolox, 2024-02-23:
> > This happens all the time with me. The engine don't take the organic supports into consideration. In overrall the travel engine is very dumb in prusaslicer. Time to go to cura...
> 
> Cura v5.6.0 has the exact same bug, reported on github as well since May 2023. [Ultimaker/Cura#15479](https://github.com/Ultimaker/Cura/issues/15479)
> 
> So I actually wanted to get PrusaSlicer instead and I ended up here... Seems like both using the same trave engine. The organic supports are great idea but they are useless untill this will be resolved. Every 3rd print is ruined.

In Cura you have to turn on Optimize wall printing order, Avoid printed parts when travelling, zhop only over printed parts, avoid suppot parts when travelling, and maybe are others...