prusa3d/PrusaSlicer #10078: [REQUEST] Move to XY then down to Z at print start

Merlin246, 2023-03-15:
**Is your feature request related to a problem? Please describe.**
Moving to the start XY after going to first layer height Z at travel speeds causes mesh compensation to intermittently stop the travel. Also causes any oozing to get stuck to the nozzle.

**Describe the solution you'd like**
Swap the layer change and XY positioning gcode order at the start of the print, then turn on mesh compensation.

**Current GCode**
G29 S1
.
.
.
G1 Z0.2
G1 Xxxx Yxxx

**Requested Change**
G1 Z10
G1 Xxxx Yyyy
G1 Z0.2
G29 S1
_start print_

**Describe how it would work**
See above

**Describe alternatives you've considered**
Not have a slightly oozing extruder... D;

foreachthing, 2023-03-16:
Yeah, I feel you!!!
For exactly that purpose, I've written this post-processing script: https://github.com/foreachthing/Slic3rPostProcessing.
It does just that.

Add this line (modified to your environment) to: Print Settings -> Output options -> Post-processing scripts:
`"C:\Python\python.exe" "c:\pathtoyour\cloned-dir\Slic3rPostProcessing\SPP-Python\Slic3rPostProcessor.py" --xy ;`

Before _(0919_Shape-Box)_ and after _(0920_Shape-Box)_ post-processing:
[image]

Merlin246, 2023-03-16:
@foreachthing Wow that's great, will definitely give it a go! Is it possible to add turning off mesh compensation before the XY move and then back on just before the Z move?

Also will RRF be an issue for this?

foreachthing, 2023-03-16:
> turning off mesh compensation

What do you mean?
All I do, is move existing lines around and adjust the speed towards first layer height.
You can pull my code and change it to your needs. Or maybe make a FR in my repository, not here!

> will RRF be an issue for this?

What is RRF?

Merlin246, 2023-03-16:
> What do you mean?
Mesh compensation is the movement in Z to match the uneven-ness of the bed. When travelling at high speeds it can cause jerky motion if the bed (like mine) is exceptionally not flat and has to make large corrections. However at 10mm+ there is no risk in colliding with the bed so it can be turned off for that initial movement so the motion is smoother.

> What is RRF?
RepRap Firmware - the firmware used by Duet3D control boards.

foreachthing, 2023-03-16:
My code is optimized for an Ultimaker with a flat, glass plated bed.
So, no. No meshcompensation. You'd have to come up with that yourself.
RepRap? Maybe... just try it ;-)

NyxCode, 2023-04-28:
Would also really like to see this implemented!

closed 2026-09-29 by prusaslicer-bot[bot]