prusa3d/PrusaSlicer #10081: Prusa Slicer 2.6a5 Slicing Bug

pbbft, 2023-03-16:
### Description of the bug

Tried to use paint on supports. Slicer ignores the use supports only on painted area does what it wants. I feel so used.

### Project file & How to reproduce

pick a file, paint on supports, change it from grid to organic, change it to enforces only. Slice, watch it put supports everywhere else. cry

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6a5

### Operating system

Win11

### Printer model

Prusa Mk3S+

jmd73, 2023-03-16:
@pbbft Please upload a 3mf with a project that doesn't work as expected,

I just tried it (2.6.0-alpha5) with a simple object (from the gallery) and it seems to work as expected.

[image]
[image]
[image]

yarus23, 2023-03-16:
I reported an issue like this https://github.com/prusa3d/PrusaSlicer/issues/9813
But I do not understand answer

closed 2026-09-29 by prusaslicer-bot[bot]