prusa3d/PrusaSlicer #10083: prusaslicer 2.5.1. error -47 on Mac

MixTapedk, 2023-03-16:
### Description of the bug

After downloading the new prusa slicer 2.5.1, the application cannot be started.
Mac writes error -47 and the application does not even start.
I working on apple Mac Studio (Mac OS Ventura 13.2.1)

Now I have to downgrade back to the original version because I need the application .

### Project file & How to reproduce

just open app and app write error -47 and fall down

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.1. Mac

### Operating system

Mac OS 13.2.1

### Printer model

n

lukasmatena, 2023-03-16:
We already saw reports like this in the past, e.g. https://github.com/prusa3d/PrusaSlicer/issues/8321. Are you absolutely sure you are running the correct binary? It should be this one: https://github.com/prusa3d/PrusaSlicer/releases/download/version_2.5.1/PrusaSlicer-2.5.1+MacOS-universal-202303131337.dmg.

MixTapedk, 2023-03-16:
thank you very much works perfectly

amcc, 2023-05-19:
I have the same error - I downloaded the latest 2.5.2 drivers and apps from this page for mac:
https://help.prusa3d.com/downloads?_gl=1*1ciw5kf*_ga*MjAxMzUwNDI1MS4xNjgzNjI0NTYy*_ga_3HK7B7RT5V*MTY4NDQ4OTI1NC42LjEuMTY4NDQ4OTI3MC40NC4wLjA.#_ga=2.92653469.1420424804.1684489255-2013504251.1683624562&_gac=1.251800315.1683706777.CjwKCAjwge2iBhBBEiwAfXDBRynF4mzKnko5RzM4WfIEfPzVq3QlCzanfSXBSlzEYYtaA7GGnI1nrhoCX28QAvD_BwE