prusa3d/PrusaSlicer #10084: [Bug] First layer % width setting seems off

mooklepticon, 2023-03-16:
### Description of the bug

The first layer settings (Print Settings, Advanced, Extrusion Width, First Layer) allow a width or a percent. If I put a width, it behaves as expect. For example, my default width is 0.44mm and if I put in 0.6mm, then look at the layer preview, it shows 0.6mm per the color legend on the left.  However, if I put 150% in there, it shows a width of 0.3mm.  I would expect it to be 0.66mm since 0.44*1.50=0.66.  It feels like the slicer is starting with a line width of 0.2mm. If I put 100%, then my first layer is 0.2mm.

### Project file & How to reproduce

[yafic_v2.zip](https://github.com/prusa3d/PrusaSlicer/files/10991394/yafic_v2.zip)

[image]
showing the first layer at 0.2mm

[image]
showing the extrusion width at 0.44mm and the first layer at 100%, so I'd expect 0.44mm, but the 1st screenshot shows dark blue which is 0.2mm per the color legend.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.1

### Operating system

Windows 10

### Printer model

Ender 3 original

neophyl, 2023-03-16:
Please read the tooltips.  When set to percentage based vales it is computed over first LAYER HEIGHT and NOT nozzle or default extrusion width.  So yes it is using 0.2 as a basis as that will be what your layer height is configured as.

It has always done this.  It has also been raised multiple times.

mooklepticon, 2023-03-16:
Oh, thanks! That's really counter intuitive.

Bumblebiber, 2023-03-16:
I wonder why they still haven't changed that. Especially because the code is already there (SuperSlicer).

rtyr, 2023-03-21:
Closing as a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2829.

We plan to do something about that, it is not that simple if you think about all possible consequences.

closed 2023-03-21 by rtyr