prusa3d/PrusaSlicer #10089: Automatic organic supports ignore support blockers.

Pyro-Fox, 2023-03-16:
### Description of the bug

Even though I blocked supports on a group of faces on my model, automatic supports with the organic setting isn't respecting my support blockers.

Bottom-up view into the small holes where supports are blocked
[image]

Angled view to show that the entire whole has support blocked (I used smart fill)
[image]

Bottom-up view showing there are now supports on the blocked areas.
[image]

### Project file & How to reproduce

[Prusa project for 2.6.0a5 bug.zip](https://github.com/prusa3d/PrusaSlicer/files/10995884/Prusa.project.for.2.6.0a5.bug.zip)

Repro steps are pretty clear. Try blocking supports with Automatic supports enabled, and check if supports were actually blocked or not.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha5

### Operating system

Windows 10

### Printer model

Custom FolgerTech FT-5

kubispe1, 2023-03-20:
**PrusaSlicer-2.6.0-alpha5** Confiramtion of issue
[image]

**PrusaSlicer-2.6.0-alpha5+64**
[image]

We probably fixed this issue with something else and should be fine in the next release.
I am closing this issue. If you will find some trouble after the next release. Let us know.

closed 2023-03-20 by kubispe1

Vashiru, 2023-04-01:
I'm actually still seeing this in alpha 6. **Note**: that this was painted (as well as the support blockers) added in alpha 5. I then saved it as a 3mf file, updated to alpha 6 and re-opened it, hoping this would solve this issue, but it doesn't seem like anything changed. See the examples on the ghost benchy:

[image]

[image]

And same for painted:

[image]

[image]

[image]

Project file:
[Ghostship_bency.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11127978/Ghostship_bency.3mf.zip)