prusa3d/PrusaSlicer #10100: Monotonic version of the concentric 100% infill pattern and/or top/bottom solid infill pattern

mkoistinen, 2023-03-18:
**Describe the solution you'd like**
I would be nice to have the concentric solid infill pattern equivalent of monotonic infill. The infill would continue spiraling around in the same direction to fill the area, rather than start/stopping to change directions and alternate the perimeter it is extending.

**Describe how it would work**
Currently, if you choose concentric fill pattern (for same bottom or top layers), it acts like you've set infinite perimeters and adds perimeters on the outside, then the inside in alternation of the surface until they meet in the middle. It also changes the circular direction. This is unfortunate for cases where the user wants a monotonic, but concentric surface.

Ideally, the slicer could use a spiral pattern that started on, say, the inner edge and continue in the same spiral direction until it meets the outer edge. **This would be ideal for use with silk and/or co-extruded filaments to perfect the shimmer/iridescence effect on a flat surface.**

I realize that what I'm describing would ideally work on circular parts and it isn't obvious to me how this would work on non-circular parts, but I imagine that the concentric infill suffers from the same complexity.

**Describe alternatives you've considered**
I've tried concentric infill for bottom or top layers, and unfortunately, it does the outside, then the inside in alternation and changing the clock-wise direction at each change. I've also tried a modifier to turn off the bottom or top layers altogether, then attempt to use the concentric infill at 100%, but it had the same effect. Both also leave an unsightly seam where each concentric ring starts and stops.

MakeFancy, 2024-03-19:
> I think you're describing Archimedean Chord infill, more or less. There's an issue with some of my models where it alternates direction (inside out vs. outside in) layer for layer, but otherwise it's the single continuous spiral you describe.

The archimedean chord pattern seems to only work when when the perimeter is a perfect circle, at least it is in orca slicer, meaning that if the perimeter is a square the top surface would still print as a circle starting and stopping at the corners of the square until the remaining gap is a circle. 
[image]
 I too am looking for a similar solution for a concentric top layer that doesn't skip from inside to outside. even if it isn't a 'spiral' just being able to print adjacent to an already printed wall for each path instead of skipping to the middle of the 'top surface' and back again would be helpful in so many ways

Lexicality, 2026-09-02:
The release notes for 3.0 don't indicate that this is fixed