prusa3d/PrusaSlicer #10101: Box Modifier Infil and pattern ignored if set object modifier

yardu1, 2023-03-18:
### Description of the bug

I am using as an example model downloaded from printables.com : https://www.printables.com/model/390823-model-tatry-148
the 3mf file of body has set infill pattern and density for body parts. Settings for pattern is gyroid and density is set to 15%.
I decided to improve pins on the body adding an additional box modifier to improve density of the pins, so they will be stronger. any changes of this modifiers in density and/or pattern not taking an effect in the final slicing results. It still generates gyroid pattern and only 15% of infill.

Thanks.

### Project file & How to reproduce

[TATRA 148 (kabina+korba)_bug.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11009891/TATRA.148.kabina%2Bkorba._bug.3mf.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version: 2.6.0-alpha5+MacOS-x64 Build: PrusaSlicer-2.6.0-alpha5+MacOS-x64-202303061452

### Operating system

Operating System: Macintosh System Architecture: 64 bit System Version: macOS Ventura Version 

### Printer model

prusa mks 3+

Snuff1eupagus, 2023-03-19:
@yardu1 

You have two Objects in your scene, and you applied the modifiers to the wrong Object.

yardu1, 2023-03-19:
Yes you are right.

closed 2023-03-19 by yardu1