prusa3d/PrusaSlicer #10102: [ 2.6-alpha5 ] Provide a way to disable the Vertical Shell thickness completely

meltiseugen, 2023-03-19:
**Is your feature request related to a problem? Please describe.**
I would like to be able to disable vertical shell thickness completely 

**Describe the solution you'd like**
a checkbox in one of the setting tabs that would allow switching on/off this feature

**Describe how it would work**
On = generation of vertical shells thickness (default option)
Off = no vertical shells are generated 

**Describe alternatives you've considered**
N/A

**Additional context**
when wanting to crate draft models or just simple models, vertical shells thickness just takes up time and filament.

mk-mrshll, 2023-04-01:
Absolutely agree with OP here. The new algorithm that adds concentric infill is MUCH better, but we need to bring back the toggle for "ensure vertical shell thickness" so that we can print fast "draft" quality prints when we want to.

Tupson444, 2023-04-01:
@mk-mrshll If I'm not mistaken, the "ensure vertical shell thickness" only switched between two different algorithms that both generated extra material, and one of which (depending on geometry) may have created less material than the other (yes, this is confusing and counter intuitive). However, what is asked here is to finally introduce an option to actually be able to turn off any of that kind of extra material.

mk-mrshll, 2023-04-01:
@Tupson444 it seems you are right.

From the 2.6.0 alpha 5 notes:
> The old "Ensure vertical shell thickness" parameter used to switch between the upstream Slic3r and Slic3r Prusa Edition algorithms to ensure vertical wall thickness.

However, I still totally agree with needing a way to just disable it altogether, and I think that the "Ensure vertical shell thickness" checkbox can be brought back to do just that.

knightsljx, 2023-04-02:
this problem has been around for literal YEARS

smajlySVK, 2023-09-26:
Also begging for adding checkbox to turn off "Ensure vertical shell thickness". It causes blobs on outer walls, which looks like a seam, also top layer quality is worser than with "Ensure vertical shell thickness" turned off. I am stuck on PrusaSlicer 2.6.0 alpha3, which is last version with ability to turn off the "Ensure vertical shell thickness". It would be perfect to keep infill selected by user, not disturbed with every second layer with concentric infill. I would like to use newest versions of PrusaSlicer, but quality of prints are very poor with versions 2.6.0 alpha5 and higher. I am printing cake toppers.

saveman71, 2023-11-22:
Thanks to a [workaround from @neophyl on the printables.com forums](https://forum.prusa3d.com/forum/prusaslicer/removal-of-solid-infill-on-a-part-with-zero-infill/#post-681172), I was able to remove completely the solid infill by adding a height modifier on the whole height except bottom and top, then setting these parameters

[image]

But I would vastly prefer an option to turn that off in the settings.

Tupson444, 2023-11-22:
This workaround works well, but not if the shape has horizontal surfaces on heights somewhere in between, as you would still want top/bottom solid infill there. PrusaSlicer definitely needs a proper solution in the settings.

neophyl, 2023-12-05:
That is a beautiful example of the difference that extra unneeded material makes.

Tupson444, 2023-12-12:
Even Maker's Muse has mentioned how he'd like to be able to disable this, showing another reason why this extra material is unwanted sometimes:
https://youtu.be/iWZw7RO2Sks?t=943

flowsean, 2024-01-22:
YES PLEASE! we run a 14-printer farm and lots of items in clear filament, all these extra unnecessary travels and extrusions are visible and ugly in clear parts. this is one of the only deal-breakers preventing us from using prusa slicer for most of our parts.

x5engine, 2024-05-07:
> I was able to use the [workaround](https://github.com/prusa3d/PrusaSlicer/issues/10102#issuecomment-1822361654), but it required some other tradeoffs. I would much rather be able to disable this infill entirely, since the extra plastic is functionally unnecessary — I already know the part is in fact watertight without it — and creates an unacceptable surface quality inside and out.
> 
> <img alt="Screenshot 2023-12-05 at 09 54 42" width="571" src="https://private-user-images.githubusercontent.com/99231/288098323-001c9331-bffc-4aa1-a5c5-238ea991af8e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTUwODA0ODksIm5iZiI6MTcxNTA4MDE4OSwicGF0aCI6Ii85OTIzMS8yODgwOTgzMjMtMDAxYzkzMzEtYmZmYy00YWExLWE1YzUtMjM4ZWE5OTFhZjhlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTA3VDExMDk0OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTAyYzk1M2U4YWIxZjA1ODU5ZDQ3YjM2NmY4ODgzZjdmZjllYmZlMTJkMmQyZTVmZjlhODE0ZjhjYWMzYWNiMDcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.bJzEBQaF1OoYOYd9nwAxlYZwfk4oZcDVZ603oNOzUo4">
> Lots of extra stringing and poor surface quality due to extra cooling and/or contraction?
> 
> ![IMG_1329 Large](https://private-user-images.githubusercontent.com/99231/288100607-fa519f24-7230-4a20-8e35-e3dfd202f047.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTUwODA0ODksIm5iZiI6MTcxNTA4MDE4OSwicGF0aCI6Ii85OTIzMS8yODgxMDA2MDctZmE1MTlmMjQtNzIzMC00YTIwLThlMzUtZTNkZmQyMDJmMDQ3LmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTA3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUwN1QxMTA5NDlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05ZmUyNDVmMTVhMTM5ZTY2NDk3ZTVlZmVjZWY3N2U3MTk4NTgwOTgxZjQzM2Y2NGNjNDdiYzE4YjgyMzFhOTYzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.1cqPcTY78YjhxEAQoiZz5ge_swb4VjFmHtyUUHURzCQ)
> 
> Identical settings but with a modifier setting `Bottom solid layers: 0` for everything above the base of the part, producing the desired quality that was possible using the same geometry slicing on earlier PrusaSlicer versions with the Ensure vertical shell thickness turned off:
> 
> ![IMG_1328 Large](https://private-user-images.githubusercontent.com/99231/288100658-04720b39-86b2-4514-b4ba-17ddbee3607b.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTUwODA0ODksIm5iZiI6MTcxNTA4MDE4OSwicGF0aCI6Ii85OTIzMS8yODgxMDA2NTgtMDQ3MjBiMzktODZiMi00NTE0LWI0YmEtMTdkZGJlZTM2MDdiLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTA3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUwN1QxMTA5NDlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00NmU0OTE0ZjQ0NTBhMzM2N2VmMWU3YzYyNzg2ZGI3ZjhlMDQ5ZTFjOTFkMTJlMGE0YzUxNjBlODcxODI1ZmI2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.3bchk1mRfuy6Z4KrkqCwsFCZrwzkKGv2p-MvrzUglBY)

Thank you the workaround works and saved 1 hour of printing those infills thanks

314Sami, 2024-09-21:
> Thanks to a [workaround from @neophyl on the printables.com forums](https://forum.prusa3d.com/forum/prusaslicer/removal-of-solid-infill-on-a-part-with-zero-infill/#post-681172), I was able to remove [...]

Thanks for the tip. In my print the time savings is about 50 % and I had to put 2 separate Ranges and Layers and Perimeters -modifiers to reach my goal. See attached 3mf. I'd really like the slicer to do this without manual work.
[4_Cell_-_18650_Bracket_v9_fixed.zip](https://github.com/user-attachments/files/17085431/4_Cell_-_18650_Bracket_v9_fixed.zip)

lukasmatena, 2024-11-21:
"Ensure vertical shell thickness" option is restored in [2.9.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1) and hopefully configurable enough to cover all use cases.  Closing.

closed 2024-11-21 by lukasmatena