prusa3d/PrusaSlicer #10103: Add Between Extrusion Role Change G-Code :)

Alexander-T-Moss, 2023-03-19:
I'm currently using SuperSlicer, but really want to migrate back to PrusaSlicer, but at the moment there is one feature of SuperSlicer I'm now reliant upon preventing me to swapping to PrusaSlicer, Between Extrusion Role Change G-Code (and custom variables to accompany it)

These are the screenshots taken out of SuperSlicer showing how it's implemented there
[image] [image]

I've looked through the sliced G-Code from PrusaSlicer, and the extrusion role markers (e.g. `;TYPE:Perimeter`) are present (like they are in SuperSlicer) So I assume most of the stuff needed to implement this is already in PrusaSlicer, just the UI allowing users to actually manipulate it in PrusaSlicer is missing.

I've considered making a post processing script but it wouldn't have as much flexibility (mainly the variables defined for each filament/print profile in the slicer).

If it's relevant, here is an example of the code in the `Between extrusion role change gcode` section of one of my SuperSlicer profiles. In short, the way it works is if the extrusion type matches the if statement, then it will insert the block of gcode bellow that if statement into the generated gcode before that extrusion role is printed.

```py
{if extrusion_role=~/Perimeter/}
SET_PRESSURE_ADVANCE ADVANCE={Perimeter_PA}
SET_VELOCITY_LIMIT SQUARE_CORNER_VELOCITY={Perimeter_SCV}
M221 S{Perimeter_Flow}
SET_LOGO_LED RED=1 GREEN=1

{elsif extrusion_role=~/ExternalPerimeter/}
SET_PRESSURE_ADVANCE ADVANCE={Perimeter_PA}
SET_VELOCITY_LIMIT SQUARE_CORNER_VELOCITY={ExternalPerimeter_SCV}
M221 S{Perimeter_Flow}
SET_LOGO_LED RED=1 GREEN=0.482

{elsif extrusion_role=~/OverhangPerimeter/}
SET_PRESSURE_ADVANCE ADVANCE={Perimeter_PA}
SET_VELOCITY_LIMIT SQUARE_CORNER_VELOCITY={Perimeter_SCV}
M221 S{Bridge_Flow}
SET_LOGO_LED BLUE=1

{elsif extrusion_role=~/InternalInfill/}
SET_PRESSURE_ADVANCE ADVANCE={Infill_PA}
SET_VELOCITY_LIMIT SQUARE_CORNER_VELOCITY={Infill_SCV}
M221 S{Infill_Flow}
SET_LOGO_LED RED=0.620 GREEN=0.106

{elsif extrusion_role=~/SolidInfill/}
SET_PRESSURE_ADVANCE ADVANCE={Infill_PA}
SET_VELOCITY_LIMIT SQUARE_CORNER_VELOCITY={Infill_SCV}
M221 S{Infill_Flow}
SET_LOGO_LED RED=1 BLUE=0.749

{elsif extrusion_role=~/TopSolidInfill/}
SET_PRESSURE_ADVANCE ADVANCE={TopInfill_PA}
SET_VELOCITY_LIMIT SQUARE_CORNER_VELOCITY={Infill_SCV}
M221 S{TopInfill_Flow}
SET_LOGO_LED RED=1

{elsif extrusion_role=~/BridgeInfill/}
SET_PRESSURE_ADVANCE ADVANCE={Infill_PA}
SET_VELOCITY_LIMIT SQUARE_CORNER_VELOCITY={Infill_SCV}
M221 S{Bridge_Flow}
SET_LOGO_LED RED=0.259 GREEN=0.153 BLUE=0.859

{elsif extrusion_role=~/InternalBridgeInfill/}
SET_PRESSURE_ADVANCE ADVANCE={Infill_PA}
SET_VELOCITY_LIMIT SQUARE_CORNER_VELOCITY={Infill_SCV}
M221 S{Bridge_Flow}
SET_LOGO_LED RED=0.282 GREEN= BLUE=0.314

{elsif extrusion_role=~/Thin Wall/}
SET_PRESSURE_ADVANCE ADVANCE={Perimeter_PA}
SET_VELOCITY_LIMIT SQUARE_CORNER_VELOCITY={Perimeter_SCV}
M221 S{Perimeter_Flow}
SET_LOGO_LED RED=1 GREEN=0.482

{elsif extrusion_role=~/GapFill/}
SET_PRESSURE_ADVANCE ADVANCE={Perimeter_PA}
SET_VELOCITY_LIMIT SQUARE_CORNER_VELOCITY={Perimeter_SCV}
M221 S{Perimeter_Flow}
SET_LOGO_LED WHITE=1

{elsif extrusion_role=~/Skirt/}
SET_PRESSURE_ADVANCE ADVANCE={Perimeter_PA}
SET_VELOCITY_LIMIT SQUARE_CORNER_VELOCITY={Perimeter_SCV}
M221 S{Perimeter_Flow}
SET_LOGO_LED RED=0.322 GREEN=0.086

{else}
M221 S100

{endif}
```

And here's an example of how variables are defined

```py
; Custom PA Values (n/a)
Infill_PA = 0.035
TopInfill_PA = 0.042
Perimeter_PA = 0.040

; Custom SCV Values (mm/s)
Infill_SCV = 15
Perimeter_SCV = 8
ExternalPerimeter_SCV = 3

; Custom Flow Values (%)
Infill_Flow = 102
TopInfill_Flow = 98
Perimeter_Flow = 100
Bridge_Flow = 85
```

lukasmatena, 2023-03-19:
> I've considered making a post processing script but it wouldn't have as much flexibility (mainly the variables defined for each filament/print profile in the slicer).

The configuration options are available as environment variables,as described in https://help.prusa3d.com/article/post-processing-scripts_283913

Another option that you might use are the G-code substitutions:
https://help.prusa3d.com/article/g-code-substitutions_301694

MartijnGevaert, 2023-12-14:
A major difference here is that SuperSlicer allows this per filament. This for example allows one to have a different linear/pressure advance value for different areas of the print.
Having to set this up for each print speed profile is a pain, and is much more handy to do from the filament settings.
Would also solve this issue:
https://github.com/prusa3d/PrusaSlicer/issues/11870

closed 2026-09-29 by prusaslicer-bot[bot]