prusa3d/PrusaSlicer #10105: 2.6 alpha 5 fails on this model after a cut

gjpc, 2023-03-19:
### Description of the bug

the slicer hangs on this stl file after a cut and then a slice command

I expected 4 objects 2 of the ruler and 2 connectors

slicer hangs and there are no connectors

### Project file & How to reproduce

Load the stl file
cut so the two pieces fit and have separate connectors
slice

the program hangs

 
[image]
[slicer 3.6 alpha 5 bug in-cm T square (Meshed).3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11010836/slicer.3.6.alpha.5.bug.in-cm.T.square.Meshed.3mf.zip)
[in-cm T square (Meshed).zip](https://github.com/prusa3d/PrusaSlicer/files/11010837/in-cm.T.square.Meshed.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha5+win64

### Operating system

win 11

### Printer model

mk3s+

kubispe1, 2023-03-21:
Hi, we will try to investigate your issue a bit. From a quick view, there could be some trouble with the new "Ensure Vertical Wall Thickness", but that's only a quess. 
- `after complete slice slicer 3.6 alpha 5 bug in-cm T square (Meshed).3mf.zip` this model freeze for Linux PS

kubispe1, 2023-03-22:
According to the developer's updated info older approach to generating the bridge over sparse infill caused this issue. This part was updated and tested with PrusaSlicer-2.6.0-alpha5+166 and looks ok.
In case you will find some issues after the next release let us know. 
I am closing this issue. 
Thnaks

closed 2023-03-22 by kubispe1

gjpc, 2023-04-04:
I downloaded the alpha 5 release and confirm that archery bow square now slices. When I click the one layer mode icon all the partial partial pockets, the numerals and the reticles, become pockets that penetrate through all the layers of the ruler. The print bed is seen through the outline of the numbers and reticles. When I move the vertical slider the features do not appear to penetrate all the layers. All layers are penetrated in the platter preview only when the slider is at either of its extremes. When not in single layer mode moving the slider shows the preview as expected.   Is this by design? Should I submit a new bug?
[image]