prusa3d/PrusaSlicer #10106: 2.6 Bottom layer Speed Bug

Lt-Kammello, 2023-03-19:
### Description of the bug

Bottom supported layers derives speed from maximum travel speed resulting in crazy hi prining/volumetric speed.

[image]

### Project file & How to reproduce

[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6

### Operating system

Windows 64 bit

### Printer model

Custom

rtyr, 2023-03-19:
Please provide your 3MF project file.

Lt-Kammello, 2023-03-21:
> Please provide your 3MF project file.
>> Done

[Solido10 (1).zip](https://github.com/prusa3d/PrusaSlicer/files/11029770/Solido10.1.zip)

closed 2026-09-29 by prusaslicer-bot[bot]