prusa3d/PrusaSlicer #10111: Allow the Organic Support number of shells to be manually controlled, and default to 1 if you have a 0.6+ nozzle

salamanders, 2023-03-20:
Organic supports are great!  They seem to default to a fixed number of shells (2) which is perfect for a 0.4 nozzle.  I'm using a 0.6 nozzle, and I'm betting that 1 shell would work pretty well.

1. Can I control the number of shells?
2. Should the default number of shells be less for higher nozzle sizes?

#9858

KrisWragg, 2023-03-22:
Agree, 2 x 0.6mm walls with PETG creates some pretty tough supports!

rlesac, 2023-03-28:
Another vote for adjustable number of shells on organic supports.
Though I get more than 2 shells with the 2.6.0-alpha5. Something like 2.1.

Most of the support tentacle is two shells, but some parts have more, please see the attache screenshot.

[image]

jukillo, 2023-05-19:
another one with .6 nozzle here. With two shells, supports are like a rock!

closed 2026-09-29 by prusaslicer-bot[bot]