prusa3d/PrusaSlicer #10112: Missing Clip-Multipart option in Alpha

KBalage, 2023-03-20:
### Description of the bug

There supposed to be a Clip-Multipart objects option Under Print Settings > Advanced, but it's missing in 2.6 Alpha5.

### Project file & How to reproduce

Print Settings > Advanced

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6 Alpha5

### Operating system

Windows

### Printer model

Prusa Mini, Ender 3S1 Plus

hamm3rs, 2023-06-23:
The parts I do have custom modeled infill patterns and require the option of turning off "clip multi parts "  I will continue to use superslicer until this gets changed. Thanks.

LucaBoschetto, 2024-12-18:
> PrusaSlicer now always behaves as if it was ON. 

Not true, unfortunately: paths now overlap. It would be really nice if the same behavior as before was reintroduced, even if still leaving the option always on.