prusa3d/PrusaSlicer #10118: Max Volumetric Speed doesn't consider Extrusion Multiplier

hackingotter, 2023-03-21:
### Description of the bug

With an Extrusion Multiplier > 1.0 it is possible to generate G-Code that exceeds the Maximum Volumetric Speed (MVS)

**Expected behavior**
Maximum Volumetric Speed should be a hard limit and not exceeded since it can lead to extrusion problems. 
With high Extrusion Multipliers (e.g. when going near the limit of the extruder where extruder is close to stripping) it's not obvious why a given slice exceeds the MVS.

### Project file & How to reproduce

Steps to reproduce the behavior:
1. Set Extrusion Multiplier to something > 1.0, e.g. 3
2. Set Maximum Volumetric Flow to something low, e.g. 5
3. Slice a Model
4. In Viewer, look at Volumetric Flow Rate. Is bigger than MVS, in this example up to 15

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.1

### Operating system

macOS 11.6.3

### Printer model

Ender3

hackingotter, 2023-03-21:
Related code locations:
[Here](https://github.com/prusa3d/PrusaSlicer/blob/1873252a3085837b1628b030693d33ebbcf3a5aa/src/libslic3r/GCode.cpp#L2853) I think `path.mm3_per_mm` should be replaced by `(path.mm3_per_mm * Extruder::extrusion_multiplier())`. Fairly certain about the location, less so about how to get the extrusion_multiplier.
```
if (EXTRUDER_CONFIG(filament_max_volumetric_speed) > 0) {
        // cap speed with max_volumetric_speed anyway (even if user is not using autospeed)
        speed = std::min(
            speed,
            EXTRUDER_CONFIG(filament_max_volumetric_speed) / path.mm3_per_mm
        );
    }
```

ils15, 2023-07-14:
same problem here

[image]
[image]

when i put extrusion multiplier, max vol speed desconsider it

i set max speed to 12mm/s3

closed 2026-09-29 by prusaslicer-bot[bot]