prusa3d/PrusaSlicer #10120: Bridging angle not correctly adjusted for shortest bridging distance.

LuckyResistor, 2023-03-21:
### Description of the bug

Compared to PrusaSlicer Version 2.5.1, Version 2.6.0-alpha5 does not adjust the bridging angle correctly to generate the shortest bridging distance for a feature.

- For alpha5 it makes no difference if the "Thick Bridges" option is enabled or not.
- For 2.5.1, enabling "Thick Bridges" will create the same bad result, but the bridges are created correctly without the option.

**The algorithm to find the best Angle for each bridging segment should be independent from the bridge style (thick or thin)**

## Result for PrusaSlicer 2.5.1

- No "Thick Bridges"
- No angle set (0)

[image]

## Result for PrusaSlicer 2.6.0-alpha5

- No "Thick Bridges"
- No angle set (0)

[image]

### Project file & How to reproduce

1. Open an example project
2. Slice the project.
3. Compare the bridging layer 

[example-projects.zip](https://github.com/prusa3d/PrusaSlicer/files/11026677/example-projects.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha5+MacOS-arm64-2023

### Operating system

macOS 13.2.1

### Printer model

Original Prusa i3 MK3S+

LuckyResistor, 2023-03-27:
I think I found the problem. I think, the bridging angle from the UI is passed in radians to the engine, instead of degrees. The parameter affects the result, but the input is clearly not in degrees.

neophyl, 2023-03-27:
Yeah the radians problem is already known - https://github.com/prusa3d/PrusaSlicer/issues/9963

jvasileff, 2023-04-22:
> Compared to PrusaSlicer Version 2.5.1, Version 2.6.0-alpha5 does not adjust the bridging angle correctly to generate the shortest bridging distance for a feature.

I noticed this as well - for one of my projects, 2.5.1 produced better bridge angles than 2.6.0-alpha. But then, I noticed 2.5.x isn't always good either. For example, in the attached image, the bridge on the right is good, but for the left side (which is a mirror image), a long bridge is used.

So, 2.6.0-alpha does often produce different bridge angles than 2.5.x, but neither 2.5.x nor 2.6.0-alpha consistently produce good angles.

(ignore that supports aren't enabled in the screenshot - that's irrelevant to the bridge angle)

[image]

Jan-Soustruznik, 2023-06-21:
Problem should be fixed in final release 2.6.0

Issue was closed as solved. 

thank you, have a nice day

closed 2023-06-21 by Jan-Soustruznik