prusa3d/PrusaSlicer #10121: PrusaSlicer window opens small every time on Mac

sharpie00, 2023-03-21:
### Description of the bug

When opening PrusaSlicer, the window is always too small (about 15% of the desktop screen size).  Have to drag the corners to resize it every time and then magnify the model.
[image]

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.1+MacOS-x64

### Operating system

MacOS Ventura 13.2.1

### Printer model

MK3S+

kubispe1, 2023-03-23:
Hi, can you inform us about **display settings** of your Mac?

sharpie00, 2023-03-23:
Hi.  It seems to be resolved now.  I maximized the window, closed the application and then re-opened.  It opens windowed (instead of full screen), but does take up the majority of my display.  Closing the issue.  Thanks.

closed 2023-03-23 by sharpie00

hamdeli, 2023-04-02:
This happens to me on a Windows (10) PC and following @sharpie00's comment, it will 'fix' itself. (Go full screen, close, then re-open).

I believe this happened after the slicer crashed.
My Display settings are:
2560x1440

If there's a way to provide a crash dump, I will.