prusa3d/PrusaSlicer #10127: Supports placed inside model

giveen, 2023-03-21:
### Description of the bug

Version: PrusaSlicer 2.6.0-alpha5

When placing two or models on the bed, near each other, organic supports can sometimes be placed inside the model.

[image]
[image]

### Project file & How to reproduce

[BODY_RIGHT.zip](https://github.com/prusa3d/PrusaSlicer/files/11031769/BODY_RIGHT.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha5

### Operating system

Windows 11

### Printer model

Ender 3

foreachthing, 2023-03-21:
#316 
You have to merge your objects first.

giveen, 2023-03-21:
> #316 You have to merge your objects first.

As each object has seperate printed parts that go inside it, merging would prevent me from placing the objects inside it.

coolxm, 2023-03-22:
Found the same thing, its easily fixed by moving the objects further apart but it would be nice if this is fixed
- using alpha2.0

KilianB, 2023-03-22:
https://github.com/prusa3d/PrusaSlicer/issues/9943#issuecomment-1456532985

> PS has always done this. Its expected operation. Its not just support too. Its just more noticeable now with organic support being placed so far outside the objects. Each object is sliced independently which is why you can have different settings for different objects. If you place 2 objects on top of each other the same occurs and you will have a bad day too.

> You the user are expected to place objects far enough apart to avoid this. If you need to stop it then you are expected to merge them into a single object.

While the contributors are aware of this and it will not be changed in the future the numerous tickets show that this is not expected behaviour for the user. For each of the 20+ tickets there are, there are probably 100 people not reporting this issue on github.

I would suggest that adding a small notification if the hull of 2 sliced models overlap with a link to an explanation that it can be solved by merging objects or placing them further apart would save a lot of confusion and work.

closed 2023-03-23 by kubispe1