prusa3d/PrusaSlicer #10131: A modifier with only Fuzzy skin disabled, changes the STL adding it`s perimeter to the main object

cfontes, 2023-03-22:
### Description of the bug

While trying to make text on the top of a model not Fuzzy ( while the rest was all set to be fuzzy ) I added a block modifier with fuzzy set to NONE, but when that was done the modifier left the impression of it's perimeter into the object.

What I expect was to only have that feature turned off and the modifier not to influence the STL in any other way, but it is clearly modifiying it and adding some extra lines.

### Project file & How to reproduce

[image]
[project.zip](https://github.com/prusa3d/PrusaSlicer/files/11039632/project.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha5+linux-x64-GTK3

### Operating system

Lubuntu

### Printer model

Kingroon KP3S

neophyl, 2023-03-22:
Again this is a duplicate of so many issues for example https://github.com/prusa3d/PrusaSlicer/issues/9787 (but there are many)

kubispe1, 2023-03-24:
Ok, I see there is a lot of requests.... I am closing this issue.
Thank you both

closed 2023-03-24 by kubispe1