prusa3d/PrusaSlicer #10132: Unable to slice after setting support and variable layer height to the model

eagle-ice, 2023-03-22:
### Description of the bug

[image]
Must reappear to retry 4 times
system: windows 10
versin: 2.6.0-alpha5
Steps.1. import model 2. slice 3. return add auto Layer height 4. cannot continue slicing

### Project file & How to reproduce

[_Mushroom_Lamp_5811635.zip](https://github.com/prusa3d/PrusaSlicer/files/11039565/_Mushroom_Lamp_5811635.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha5

### Operating system

windows10

### Printer model

ankermake

neophyl, 2023-03-22:
Of course you cant.  Do you not see the error message ?  Variable layer height is not supported when Organic supports are being used.  Change support type to Snug or grid and you can continue to use variable layer height.

It's in the release notes and the warning has been added specifically as so many people were reporting the same issue.

rtyr, 2023-03-22:
Answered, closing.

closed 2023-03-22 by rtyr

eagle-ice, 2023-03-27:
Thank you very much for your answer。