prusa3d/PrusaSlicer #10135: Slicer does not display color selection screen when adding in color change.

diablo1129, 2023-03-22:
### Description of the bug

with MMU printer selected trying to do a color change, the normal box that pops up to allow you to pick what color its changing to does not pop up. you can not click anywhere in the application, however if you hit the enter key, it will move on as if you hit the ok button. It still adds the color change but does not show that box which is important for what i do. 

### Project file & How to reproduce

[escape Boat magnet.zip](https://github.com/prusa3d/PrusaSlicer/files/11043545/escape.Boat.magnet.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0 alpha5

### Operating system

windows 11

### Printer model

i3 mk3s w/ mmu2s

kubispe1, 2023-03-24:
Excuse me, I don't get your issue's description.
Everything related to the MMU painting gizmo works ok in my case or maybe I overlooked something. 
Can you record .gif or video? It could help us to identify what you mean.

diablo1129, 2023-03-24:
I attached a video. It has audio where I explain, first one is newest alpha
out, the one where the screen pops up is 2.5
  20230324_120402.mp4
<https://drive.google.com/file/d/1TNUMELZgy6w1tmXuvxqk5jst6hIsBvJq/view?usp=drivesdk>

l0ne5tar, 2023-04-30:
This probably the same thing I encountered.
https://github.com/prusa3d/PrusaSlicer/issues/10419

diablo1129, 2023-06-04:
Any

diablo1129, 2023-06-04:
Sorry im not super familiar with Github didnt mean to close it. Any updates or clarification needed on this? I see its still open with no other activity so i just wanted to be sure. Thanks!

closed 2026-09-29 by prusaslicer-bot[bot]