prusa3d/PrusaSlicer #10136: Attempting to slice a model takes down my entire DE

kozross, 2023-03-23:
### Description of the bug

When I attempt to add [this model](https://github.com/rolohaun/Rook-180/blob/main/STL/Frame/Top%20Frame/TOP_Left_Front_Frame.stl) in the alpha, it ends up crashing my entire desktop environment. It takes down both monitors, as well as audio, for about 30 seconds, before it gets restarted by my OS.

This seems to happen (as far as I can tell) only on the model I linked, and not on anything else.

`sudo dmesg` produces this:

```
[   63.912461] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* ring gfx_0.0.0 timeout, signaled seq=10528, emitted seq=10531
[   63.912745] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* Process information: process slic3r_main pid 3375 thread prusa-slic:cs0 pid 3416
[   63.912995] amdgpu 0000:0a:00.0: amdgpu: GPU reset begin!
[   64.176351] amdgpu 0000:0a:00.0: amdgpu: BACO reset
[   67.320098] amdgpu 0000:0a:00.0: amdgpu: GPU reset succeeded, trying to resume
[   67.320252] [drm] PCIE GART of 512M enabled (table at 0x0000008000300000).
[   67.320318] [drm] VRAM is lost due to GPU reset!
[   67.320319] [drm] PSP is resuming...
[   67.366197] [drm] reserve 0x900000 from 0x81fd000000 for PSP TMR
[   67.408749] amdgpu 0000:0a:00.0: amdgpu: RAS: optional ras ta ucode is not available
[   67.414695] amdgpu 0000:0a:00.0: amdgpu: RAP: optional rap ta ucode is not available
[   67.414696] amdgpu 0000:0a:00.0: amdgpu: SECUREDISPLAY: securedisplay ta ucode is not available
[   67.414698] amdgpu 0000:0a:00.0: amdgpu: SMU is resuming...
[   67.414737] amdgpu 0000:0a:00.0: amdgpu: use vbios provided pptable
[   67.414738] amdgpu 0000:0a:00.0: amdgpu: smc_dpm_info table revision(format.content): 4.5
[   67.417660] amdgpu 0000:0a:00.0: amdgpu: SMU is resumed successfully!
[   67.565956] [drm] kiq ring mec 2 pipe 1 q 0
[   67.567419] [drm] VCN decode and encode initialized successfully(under DPG Mode).
[   67.567620] [drm] JPEG decode initialized successfully.
[   67.567636] amdgpu 0000:0a:00.0: amdgpu: ring gfx_0.0.0 uses VM inv eng 0 on hub 0
[   67.567637] amdgpu 0000:0a:00.0: amdgpu: ring comp_1.0.0 uses VM inv eng 1 on hub 0
[   67.567638] amdgpu 0000:0a:00.0: amdgpu: ring comp_1.1.0 uses VM inv eng 4 on hub 0
[   67.567639] amdgpu 0000:0a:00.0: amdgpu: ring comp_1.2.0 uses VM inv eng 5 on hub 0
[   67.567639] amdgpu 0000:0a:00.0: amdgpu: ring comp_1.3.0 uses VM inv eng 6 on hub 0
[   67.567640] amdgpu 0000:0a:00.0: amdgpu: ring comp_1.0.1 uses VM inv eng 7 on hub 0
[   67.567640] amdgpu 0000:0a:00.0: amdgpu: ring comp_1.1.1 uses VM inv eng 8 on hub 0
[   67.567641] amdgpu 0000:0a:00.0: amdgpu: ring comp_1.2.1 uses VM inv eng 9 on hub 0
[   67.567641] amdgpu 0000:0a:00.0: amdgpu: ring comp_1.3.1 uses VM inv eng 10 on hub 0
[   67.567642] amdgpu 0000:0a:00.0: amdgpu: ring kiq_2.1.0 uses VM inv eng 11 on hub 0
[   67.567642] amdgpu 0000:0a:00.0: amdgpu: ring sdma0 uses VM inv eng 12 on hub 0
[   67.567643] amdgpu 0000:0a:00.0: amdgpu: ring sdma1 uses VM inv eng 13 on hub 0
[   67.567643] amdgpu 0000:0a:00.0: amdgpu: ring vcn_dec uses VM inv eng 0 on hub 1
[   67.567644] amdgpu 0000:0a:00.0: amdgpu: ring vcn_enc0 uses VM inv eng 1 on hub 1
[   67.567644] amdgpu 0000:0a:00.0: amdgpu: ring vcn_enc1 uses VM inv eng 4 on hub 1
[   67.567645] amdgpu 0000:0a:00.0: amdgpu: ring jpeg_dec uses VM inv eng 5 on hub 1
[   67.570244] amdgpu 0000:0a:00.0: amdgpu: recover vram bo from shadow start
[   67.570275] amdgpu 0000:0a:00.0: amdgpu: recover vram bo from shadow done
[   67.570285] amdgpu 0000:0a:00.0: amdgpu: GPU reset(2) succeeded!
[   67.570304] [drm] Skip scheduling IBs!
[   67.570318] [drm] Skip scheduling IBs!
[   67.570328] [drm] Skip scheduling IBs!
[   67.570333] [drm] Skip scheduling IBs!
[   67.570338] [drm] Skip scheduling IBs!
[   67.570345] [drm] Skip scheduling IBs!
[   67.570350] [drm] Skip scheduling IBs!
[   67.570353] [drm] Skip scheduling IBs!
[   67.570359] [drm] Skip scheduling IBs!
[   67.570365] [drm] Skip scheduling IBs!
[   67.570368] [drm] Skip scheduling IBs!
[   67.570369] [drm] Skip scheduling IBs!
[   67.570371] [drm] Skip scheduling IBs!
[   67.570374] [drm] Skip scheduling IBs!
[   67.570377] [drm] Skip scheduling IBs!
[   67.570379] [drm] Skip scheduling IBs!
[   67.570380] [drm] Skip scheduling IBs!
[   67.570383] [drm] Skip scheduling IBs!
[   67.570387] [drm] Skip scheduling IBs!
[   67.570391] [drm] Skip scheduling IBs!
[   67.570394] [drm] Skip scheduling IBs!
[   67.570397] [drm] Skip scheduling IBs!
[   67.570399] [drm] Skip scheduling IBs!
[   67.570402] [drm] Skip scheduling IBs!
[   67.570405] [drm] Skip scheduling IBs!
[   67.570417] [drm] Skip scheduling IBs!
[   67.570424] [drm] Skip scheduling IBs!
[   67.570428] [drm] Skip scheduling IBs!
[   67.570431] [drm] Skip scheduling IBs!
[   67.570434] [drm] Skip scheduling IBs!
[   67.570437] [drm] Skip scheduling IBs!
[   67.570439] [drm] Skip scheduling IBs!
[   67.570441] [drm] Skip scheduling IBs!
[   67.570444] [drm] Skip scheduling IBs!
[   67.570446] [drm] Skip scheduling IBs!
[   67.570448] [drm] Skip scheduling IBs!
[   67.570451] [drm] Skip scheduling IBs!
[   67.570453] [drm] Skip scheduling IBs!
[   67.570456] [drm] Skip scheduling IBs!
[   67.570458] [drm] Skip scheduling IBs!
[   67.570461] [drm] Skip scheduling IBs!
[   67.570464] [drm] Skip scheduling IBs!
[   67.570466] [drm] Skip scheduling IBs!
[   67.570468] [drm] Skip scheduling IBs!
[   67.570470] [drm] Skip scheduling IBs!
[   67.570477] [drm] Skip scheduling IBs!
[   67.570483] [drm] Skip scheduling IBs!
[   67.570486] [drm] Skip scheduling IBs!
[   67.570487] [drm] Skip scheduling IBs!
[   67.570489] [drm] Skip scheduling IBs!
[   67.570490] [drm] Skip scheduling IBs!
[   67.570492] [drm] Skip scheduling IBs!
[   67.570493] [drm] Skip scheduling IBs!
[   67.570495] [drm] Skip scheduling IBs!
[   67.570496] [drm] Skip scheduling IBs!
[   67.570498] [drm] Skip scheduling IBs!
[   67.570500] [drm] Skip scheduling IBs!
[   67.570501] [drm] Skip scheduling IBs!
[   67.570504] [drm] Skip scheduling IBs!
[   67.570505] [drm] Skip scheduling IBs!
[   67.570507] [drm] Skip scheduling IBs!
[   67.570513] [drm] Skip scheduling IBs!
[   67.570515] [drm] Skip scheduling IBs!
[   67.570517] [drm] Skip scheduling IBs!
[   67.570520] [drm] Skip scheduling IBs!
[   67.570521] [drm] Skip scheduling IBs!
[   67.570523] [drm] Skip scheduling IBs!
[   67.570525] [drm] Skip scheduling IBs!
[   67.570527] [drm] Skip scheduling IBs!
[   67.570529] [drm] Skip scheduling IBs!
[   67.570534] [drm] Skip scheduling IBs!
[   67.570538] [drm] Skip scheduling IBs!
[   67.570540] [drm] Skip scheduling IBs!
[   67.570543] [drm] Skip scheduling IBs!
[   67.570545] [drm] Skip scheduling IBs!
[   67.638513] gnome-shell[1145]: segfault at 0 ip 00007f75e6ea671e sp 00007ffe8652d7c0 error 4 in libmozjs-102.so[7f75e6ea0000+acf000] likely on CPU 19 (core 9, socket 0)

```

### Project file & How to reproduce

[Empty project before import](https://github.com/prusa3d/PrusaSlicer/files/11045754/de-crash.zip)

1. Load the project.
2. Attempt to add [this model](https://github.com/rolohaun/Rook-180/blob/main/STL/Frame/Top%20Frame/TOP_Left_Front_Frame.stl)
3. Watch the DE fail hilariously.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha5+linux-x64-GTK3

### Operating system

Arch Linux

### Printer model

Prusa MINI+

kubispe1, 2023-03-24:
Hi @kozross, we have tried to reproduce it on 
- Linux Ubuntu 
- Arch Linux  (VM) 
and we are not able to reproduce it.

kozross, 2023-03-27:
@kubispe1 - seems that the recent kernel driver for AMD's graphics does _not_ like something. I downgraded to kernel 6.1.5 and it all magically works again. Since this seems like a more general problem, gonna close.

closed 2023-03-27 by kozross