prusa3d/PrusaSlicer #10139: [2.6.0alpha5] One of the solid infills missing in the lowest top solid infill layer

Ohmarinus, 2023-03-23:
### Description of the bug

The problem is that for example setting `solid layers` to 5, it will only print the top 4 layers as solid, and doesn't completely do a solid infill for the 1st top solid layer, counting from the bottom.

So a print with 25 layers, with a setting for 5 solid top layers should have a solid infill in layer 20, 21, 23, 24 and 25. Instead, it doesn't completely infill layer 20.

This happens on multiple models, but not on all models. It might have to do with the size of the hole it has to cover.

Some examples:

The calibration cube (files can be found in the project file section of this report):
[image]

Settings for 3 solid layers, the bottom one of the three top layers is only partially infilled:
[image]

[image]

Singled out the layer that is misbehaving:
[image]

The same with this file (files can be found in the project file section of this report) where the herringbone gears are not being infilled correctly on the 5th later:
[image]

[image]

The single layer that's misbehaving:
[image]

Single layer with the one below, showing the normal infill of the layer below the misbehaving layer is normal, as seen through the faulty layer that's missing infill:
[image]

The infill shows up **during** generating:
[image]

But then disappears when generating is finished:
[image]

### Project file & How to reproduce

[heringbone_calicube_infill_bug_PS260a5.zip](https://github.com/prusa3d/PrusaSlicer/files/11055812/heringbone_calicube_infill_bug_PS260a5.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha5+win64

### Operating system

Windows 10

### Printer model

DIY Delta 220mm diameter, 320mm height running RRF

foreachthing, 2023-03-24:
I just opened your two projects and I don't see any of your issues you're having.
The gears are filled and the Z is not missing anything:

[image]

[image]

I am using Version 2.6.0-alpha5+win64 as well and _OPENED_ your project. Hence, using your unmodified settings.
I don't know what to tell you ... maybe reslice it? Fix the geometry through netfabb, even if not required.

neophyl, 2023-03-24:
The has to be differences in settings between the saved and opened project that Foreachthing is opening and the profiles stored for the poster as another obvious difference is the slice gap closing, the OP gears model is merging the gears into the outer indicating a too large slice gap closing for that model and the foreachthing pics show a clear separation.  

The other thing to note is the OP pic of the missing layer on the gears is the bridge infill layer.  
[image]

Make sure you havent deselected show bridge infill from the legend or it will not show it on the preview like so -
[image]

it will show it on the initial slice though just not on the preview if  turned off in the legend.

BTW I also don't see any problems with missing extrusions in the saved projects.  Win 10 and alpha 5

foreachthing, 2023-03-24:
> The other thing to note is the OP pic of the missing layer on the gears is the bridge infill layer.

Oh, nice catch! 
I didn't see that ... now it's clear that the OP has DISABLED (clicked on) "Bridge infill" in the legend (top left).

Ohmarinus, 2023-03-24:
This is very interesting as I had not disabled that setting. Without closing the program, I was unable to turn it back on as well. I found out that after restarting PS, bridge infill was once again shown and I can now enable and disable this in the legend successfully.

So the solution was to close & reopen the application. I'm ashamed for making such a rookie mistake of not closing and reopening the program. I'm sorry.

Ohmarinus, 2023-03-24:
> @Ohmarinus, I did not check the whole conversation, but is this issue resolved?

Absolutely, I thought I had closed it, I must have pressed 'comment' instead of 'close with comment'. Sorry!

closed 2023-03-24 by Ohmarinus

kubispe1, 2023-03-27:
Don't worry, thanks