prusa3d/PrusaSlicer #10142: improve scaling

SphaeroX, 2023-03-24:
I would like to submit a proposal for improving the scaling tool in Prusa Slicer. It would be great if the tool could allow users to stretch parts in length while retaining certain geometries such as a circular cutout.

Specifically, I envision users being able to select an object and then stretch it in length by simply entering or dragging the appropriate scaling factor. At the same time, certain geometric features such as a circular cutout should retain their shape and proportions to ensure that the modified object remains functional and compatible with other parts.

I believe that such a feature would significantly ease the customization of objects to individual needs and requirements and thus enhance the usefulness of Prusa Slicer as a 3D printing software.

[image]
[image]

neophyl, 2023-03-24:
In effect this a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/7563  while the suggested process is slightly different the final outcome would be the same.

This is also easily achievable in CAD software so seems like a waste of effort doing the same thing in the slicer.

SphaeroX, 2023-03-24:
the same could be said about the new cutting tool 😉

neophyl, 2023-03-24:
Yes it could, and I did say so :)

I still use cad to cut my models and add in registration holes etc as it does a better job of it.  Still cant cut part of an object with the new cut tool for example.  Or change the angle of the registration holes relative to the cutting plane. 
Plenty of open issues regarding it too.  
I am a firm believer that the slicer should be the best slicer it can be and not a poor bit of cad software tacked onto an above average slicer.  

Now if Prusa had the resources to do ALL the jobs well then more power to them.  Unfortunately that is rarely if ever the case with any project.

SphaeroX, 2023-03-24:
I believe that the target audience for Prusa printers desires such functions as productivity factors. It is always important to consider which target audience one is addressing, which is why I think that features for CAD editing in the slicer also make sense.

kubispe1, 2023-03-24:
As @neophyl mentioned, your feature request is duplicate #7563. Before some days also @murk-sy reported nicely the same FR here #10114 . Closing but Thanks
Have a nice day

closed 2023-03-24 by kubispe1