prusa3d/PrusaSlicer #10147: Single extruder Multi Material setup produces G1 commands with negative F values (`G1 F-2147483648`)

HitLuca, 2023-03-24:
### Description of the bug

Trying to print a multicolor object, using a virtual mmu setup (manual filament changes using a single extruder): right before calling the M600 command, the gcode has a number of `G1 F-###` values, which fail the print. Relates to this SuperSlicer issue, https://github.com/supermerill/SuperSlicer/issues/3314 which seems to be present in PrusaSlicer as well

### Project file & How to reproduce

[screwmeasuringtool_v1_with_m2.5_raised_letters.zip](https://github.com/prusa3d/PrusaSlicer/files/11066847/screwmeasuringtool_v1_with_m2.5_raised_letters.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

 PrusaSlicer-2.5.0+win64-202209060714

### Operating system

Windows 10

### Printer model

Customized Eryone Thinker SE

HitLuca, 2023-03-25:
Links to https://github.com/supermerill/SuperSlicer/issues/3314

HitLuca, 2023-04-22:
Looks like that changing the `Cooling tube length` in `Printer Settings` > `Single extruder MM setup` from `0` to a very small value like `0.00001` fixes the issue until we get a proper patch

MaikelKelder, 2023-08-07:
> Looks like that changing the `Cooling tube length` in `Printer Settings` > `Single extruder MM setup` from `0` to a very small value like `0.00001` fixes the issue until we get a proper patch

This doesnt seem to work, it was 41 for me, changed it, but F-2147483648 still shows up in gcode after slicing. Im on latest version beta, since it fixes the "extruder naming" in the gcode.

MaikelKelder, 2023-08-07:
Only thing that works is add a G-Code substitution and replace the F-2147483648 with nothing. Tested and works. I can finally print again...

LostFool, 2023-12-14:
To me it seems the issue comes from the "Toolchange parameters with single extruder MM printers" in filament settings. I had one that still had settings while all others were set to 0. Can't verify if this is the issue but i toggled my changes on and off in a number of places. After fixing the one extruder that was wrong, it seemed to be good.

LostFool, 2023-12-14:
Yup. I just turned one of the filaments back on with default parameters while my other filaments are set to 0 (letting Happy Hare control it) and the F-# came back.

closed 2026-09-29 by prusaslicer-bot[bot]