prusa3d/PrusaSlicer #10148: Organic Supports Running through Object to get to another object

plundrigan, 2023-03-24:
### Description of the bug

I have had a couple of instances with multiple large/tall items in slicer, when printed have the organic support printing through one object to get to the point on the other. I do not recall seeing this behavior in the slicer prior to export. However, I will double check. 

I would not expect the supports to run through one object to get to another when slicing multiple objects on the build plate. 

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6- Alpha 5 mac os x64

### Operating system

Mac OS 12.6.3

### Printer model

Ender 3 Pro

TDA-Holdings, 2023-04-17:
I have the same problem when objects are close together on build plate on MK3S+.

Version of PrusaSlicer

2.6- Alpha 6 mac os x64

Operating system

Mac OS 10.15.7 (Catalina)

Printer model

Prusa MK3S+

neophyl, 2023-04-18:
https://github.com/prusa3d/PrusaSlicer/issues/10032 or https://github.com/prusa3d/PrusaSlicer/issues/316 or https://github.com/prusa3d/PrusaSlicer/issues/8947 (which itself has a list of 8 previous issues that was a duplicate of).

This is reported all the time.  Please search for both open and Closed issues before opening new ones.

closed 2026-09-29 by prusaslicer-bot[bot]