prusa3d/PrusaSlicer #10149: Firmware crash AGAIN

Cmos2112, 2023-03-24:
### Description of the bug

I had a perfectly functional MK3S+ with MMU2 working just fine but asking me to upgrade the firmware every time I started a new print. So I updated both the printer and the MMU. I expected it to work. I have fought every single thing on this printer since the upgrade to the new firmware. I have reset the mmu to zero and recalibrated the bowden tube for all 5 filaments as per instructions. I have shut down the thermal whosawhatsit because it is not compatible with E3D Revo. I have been really patient all the while and now I want to burn this sob down to the ground and buy an X1C to save myself the royal pain in the posterior. I have not made huge dramatic changes to this printer, and in fact the changes I had made were working great on the old firmware.  So the addition of the Revo and the change to a gear reduction extruder aside this is otherwise stock. so why does the mmu not feed a filament to the extruder during a print now?
I had to change some things in the print profile to make the slicer recognize my filament profiles and then sneak a .8 nozzle size in place of the .6 stock profile for the mmu setup. So maybe that is where the issue resides. But why would a functioning profile not turn the motor to load filament to the extruder. I am certain I did not turn off the feed motor for the mmu with any setting in the slicer. The selector moves successfully to the filament I want to use, but it will not feed it out. The printer says the firmware crashed and to contact help for support but i ended up here because I am too dumb to ask the right question in the right format at the right time. 
I don't know how to get the debug file off the printer and I don't know why I ended up here and not somewhere that can help me. So read this rant and delete it if you must and I will go on trying in vain to repair what NEVER SHOULD HAVE BROKEN IN THE FIRST PLACE!
PS if you made it through my obvious frustration and haven't written me off as a lunatic already, please offer your suggestions and advice. If I find some way to ever get back to this report and see what you had to tell me it will be a miracle, but I do appreciate the effort on your part random human.

### Project file & How to reproduce

I have no idea how to do this at all.
I am not a programmer I just want the thing to do the thing it's supposed to do like it did without incident 2 weeks ago.
Also I am sorry that I don't seem very helpful in trying to get this resolved. I am mad enough to light my hair on fire right now. I should not be on the internet ranting but I am anyway.
Please don't block me forever, just laugh at the dumb guy and move on or offer help.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.1

### Operating system

windows 11

### Printer model

MK3S+

neophyl, 2023-03-25:
You want the Mk3 github- https://github.com/prusa3d/Prusa-Firmware/issues  this is the prusa slicer github project.

The best advice to those having issues with the 3.12.x firmware is to roll back to the previous versions.

Cmos2112, 2023-03-25:
Neophil, thanks for that. I did realize that as I was finishing my rant. I was just so ticked off I couldn't think straight at the moment. Caffeine imbalance or mental imbalance or whatever. 
Time to make like 90's walmart and do a Rollback.

neophyl, 2023-03-25:
You can turn the nag off though, if you look in the start gcode field in Prusa Slicer>Printer Settings.>Custom Gcode youi should see a line like the following -
M115 U3.12.2 ; tell printer latest fw version, 
just comment out that line and save it as a custom printer profile.  Then when you slice with that 'new' printer it wont do the firmware compare.