prusa3d/PrusaSlicer #10150: prusa 2.6.0. a5 crash

Thebears75, 2023-03-25:
### Description of the bug

during the slicing of a small modem i get a message of insufficient memory but the system has more than sufficient availble.
after that the software keep on crash

### Project file & How to reproduce

 
[Aln448_porta_Motore.zip](https://github.com/prusa3d/PrusaSlicer/files/11068287/Aln448_porta_Motore.zip)
import 
fix with netfabb 
slice
at this point tha app will crash

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0.alpha 5 win64

### Operating system

Windows 11 home

### Printer model

prusa M3Ks+

kubispe1, 2023-03-27:
Thanks, we will look into it.
SPE-1612

Thebears75, 2023-04-23:
hi i tried the same with the alpha6 same result

FidelCapo, 2023-05-02:
This issue should be fixed and fix will be part of next release.
Closing.

closed 2023-05-02 by FidelCapo