prusa3d/PrusaSlicer #10151: Nozzle scar on external perimeter while arachne + avoid crossing perimeters

Moyrn, 2023-03-25:
### Description of the bug

Context : Printing a tube-like model, no retractation, avoid crossing perimeters
- I opened an "old" 3mf file which was printing fine, now with 2.5.0/arachne, and the print leaves bump scars on the external perimeter.
- The shape is pre-echoing an oval shape to be printed further
Photo of the bumps/scars
[image]

- looking at the moves in prusa slicer, i see these are created by move angles of the nozzle while avoiding perimeters.
These moves do happen in the internal perimeter - which should be fine. But alas, although printing external perimeters later, the scar/bump is still visible. 
Explanation on the move angles
[image]

- Please note that i use the E3D revo nozzle. Maybe it happens with normal nozzle as well. This is the only issue i faced since one year of revo

Expected : no scar -- see photo below

How to avoid : remove arachne, back to classic perimeters algorithm
Photo without arachne
[image]

### Project file & How to reproduce

Project file
[PR-Mini.zip](https://github.com/prusa3d/PrusaSlicer/files/11068313/PR-Mini.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0-Win10

### Operating system

Windows 10

### Printer model

MINI

Moyrn, 2023-08-26:
still have it with 2.6.0

hejllukas, 2025-01-04:
Not reproducible in 2.9.0. Closing.

closed 2025-01-04 by hejllukas