prusa3d/PrusaSlicer #10154: reanudar impresion extrusor no inyecta manterial

tortiemilianoar, 2023-03-25:
### Description of the bug

Al pausarse la impresión y querer reanudarla la impresora realiza todos los movimientos correspondientes pero el motor del extrusor, el que inyecta material a nozzel no avanza, queda trabado. este problema me pasa solamente cuando la mino con prussaslicer, e probado con otros laminadores y no tuve problemas.
una lastima porque prusa realmente me encanta para trabajar.
existe algun comodando del gcode que pueda cambiar para solucionar este problema?

### Project file & How to reproduce

.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.1 y 2.6.0

### Operating system

windows 10

### Printer model

Artillery Genius Pro

Rosenbrock92, 2023-09-20:
Hola ya lo resolví escribíme

gasgodoy, 2023-09-28:
@Rosenbrock92 tengo el mismo problema, no sabía que solo era con prusaslicer, ya que no he usado otro laminados, siempre culpé a la impresora, no al laminador, puedes compartir con nosotros que cambiar?

Rosenbrock92, 2023-09-28:
Hola, si 
Son tres cosas que Tenes que hacer 
La primera en modo experto vas a configuración de la impresora , General (que está a la izquierda) y abajo de todo sacar el tilde en usar distancias relativas ,
Luego en código personalizado la segunda línea dice M83 y va M82. Y abajo de todo en código g antes de cambiar de capa está G92 E0 fíjense que quede ;G92 E0.
Cualqueir cosa me hablan al instagram no hay problema matiasburalli y les mando video o doy una mano.

gasgodoy, 2023-09-29:
muchas gracias, lo pruebo y te cuento como anduvo, ayer murió una de mis
printers se le rompio el termistor

gasgodoy, 2023-11-21:
Hola, como estás? recién hoy pude probar tus recomendaciones y funcionaron de lujo!!
probe de cambiar filamento, y anduvo, probe de detener y reanudar la impresión y anduvo, probé de cortarle el suministro eléctrico y anduvo, peeeerooo si se queda sin filamento no se detiene alguna sugerencia?

Rosenbrock92, 2023-12-07:
> Hola, como estás? recién hoy pude probar tus recomendaciones y funcionaron de lujo!!
> probe de cambiar filamento, y anduvo, probe de detener y reanudar la impresión y anduvo, probé de cortarle el suministro eléctrico y anduvo, peeeerooo si se queda sin filamento no se detiene alguna sugerencia?

Si hay un código que Tenes que agregar el M600.
Podes ver mis trabajos con artillery  en sithprint háblame por ahí si queres y te ayudo 
https://instagram.com/sithprint?igshid=MmVlMjlkMTBhMg==

closed 2026-09-29 by prusaslicer-bot[bot]