prusa3d/PrusaSlicer #10160: Crashing while slicing model with organic supports

d3nver2140, 2023-03-26:
### Description of the bug

I'm trying to slice one model using organic supports, i've changes overchang to 35 degree and after a while from the start slicer is disapearing

### Project file & How to reproduce

[alien-full-figure.zip](https://github.com/prusa3d/PrusaSlicer/files/11070491/alien-full-figure.zip)
Trying to slice this model, 80 precent of it's dimensions

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0 - alpha 5

### Operating system

Windows 11

### Printer model

MKS3+

EdsonMaiero, 2023-03-26:
I have the same problem when using organic supports. The app frozen at 70% and I need to re-start the prusaslicer. I am using a Imac.

kubispe1, 2023-03-27:
Can you add 3mf for correct bug detection, please ? I am not able to reproduce it. Thanks

foreachthing, 2023-03-27:
How much RAM do you have?
It sliced on my Win11 (32 GB) just fine. Even after enabling Organic Support.
BUT: I've fixed the parts with the :warning: signs first.

EdsonMaiero, 2023-03-30:
> BUT: I've fixed the parts with the ⚠️ signs first.

Hello,

I am using 64GB - MacOS Ventura 13.2.1

Edson Maiero Lins
(011)97100-1924

à(s) 15:47:

EdsonMaiero, 2023-04-01:
I am using 64GB, and that's is the 3mf file:

https://www.dropbox.com/s/vhqhheo03xzdvr6/yodafigure.3mf?dl=0

regards,

Edson Maiero Lins
(011)97100-1924

à(s) 15:47:

EdsonMaiero, 2023-04-01:
Good morning,

it is the 3mf file:

https://www.dropbox.com/s/vhqhheo03xzdvr6/yodafigure.3mf?dl=0

regards,

Edson Maiero Lins
(011)97100-1924

à(s) 06:22:

lukasmatena, 2023-04-04:
@d3nver2140 I am somewhat puzzled. You claim that the problem happens when you use organic supports and overhang threshold of 35 deg, but the 3mf that you enclosed uses grid with 12 degrees. You are claiming that you scaled the stl to 80%, but the parts in the 3mf are clearly the full size.
We will not be able to reproduce it this way. Please, set up EVERYTHING exactly so it would crash, save the project JUST before slicing, verify that clicking Slice Now indeed leads to the crash and only then enclose the project that you created. Then we might be able to fix it. Thanks.

@EdsonMaiero Your issue is very likely unrelated to this one. It is not crashing but freezing and it is already fine in 2.6.0-alpha6, which makes it a likely duplicate of https://github.com/prusa3d/PrusaSlicer/issues/10048.

closed 2026-09-29 by prusaslicer-bot[bot]