prusa3d/PrusaSlicer #10163: Config bundle cannot be exported when "idle temperature" is specified for filament

PolymerPrints, 2023-03-26:
### Description of the bug

When a filament has idle temperature specified, config bundles cannot be exported. Unchecking idle temperature on all filaments that have it enabled will allow exporting again.

[image]

### Project file & How to reproduce

[idle_temp_error.zip](https://github.com/prusa3d/PrusaSlicer/files/11071198/idle_temp_error.zip)

1. Go to File -> Export
2. Perform any export config action
3. Pop-up with error will appear

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha5

### Operating system

Windows 10

### Printer model

Snapmaker 2.0 A350

kubispe1, 2023-03-27:
Thanks, we will look into it.
SPE-1613

lukasmatena, 2023-04-02:
Fixed in [2.6.0-alpha6](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha6). Closing.

closed 2023-04-02 by lukasmatena