prusa3d/PrusaSlicer #10166: First layer is not complete

usarskyy, 2023-03-26:
### Description of the bug

Problem: Only ~80% of the first layer is printed
Expected: 100% of the first layer must be printed.

[image]

All the files are attached.

### Project file & How to reproduce

[prusa_bug.zip](https://github.com/prusa3d/PrusaSlicer/files/11071826/prusa_bug.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version: 2.5.1+win64, Build: PrusaSlicer-2.5.1+win64-202303151108

### Operating system

Windows 10 (build 22621), 64-bit edition

### Printer model

Creality Ender 6

kubispe1, 2023-03-27:
I think it is clear. I am closing it.  
Thanks

closed 2023-03-27 by kubispe1

usarskyy, 2023-03-27:
I used this tool after I turned the part by 180 degrees but now I understand that it doesn't work as I expected.

My expectation was that when a part is selected and is not touching the surface completely (for example, like in my case), the position of the selected part is automatically adjusted when I press that button. In result the nearest flat surface must lie down on the bed's surface.
But in reality the tool does nothing until I touch the surface that I want to lay on the bed's surface.

Question: as I said, I rotated the part manually. Rotation angle was **almost** 180 degrees. Is there any indicator that shows rotation angle compared to the original position?

For example, right now the only deviation I see is "0.08mm" (which is actually not completely true because the whole part is not elevated, just of the corners is):

[image]

Here is the correct placement:

[image]