prusa3d/PrusaSlicer #10168: A few small changes that I think would make Prusaslicer better (or at least make me use it as my primary)

LiterallyKey, 2023-03-26:
TLDR: I love mouse ears and is the #1 reason why I primarily prefer Superslicer over Prusaslicer. Superslicer's implementation could just be copied for a (I assume) simple way to make Prusaslicer better, but there are also some potential ways to make it even better. I would honestly be pretty happy and switch to Prusaslicer as my standard with just that change alone, the rest of what I would like to see are mostly just adding more control and quality of life features. Solid infill specific pattern and encroachment settings as well as chamber temp would also be relatively easy and more important. Most of the rest are probably too much of a change or too complex to be realistically added, so maybe it would be better to figure out how to make a fork and try them myself (which is unlikely to happen)

I apologize if this isn't the place or way to do this, but I figure there might be a small chance it could be useful to outline exactly what I personally would like to see in Prusaslicer and where I think it is limited. It turned out extremely long though. 

For context: I have primarily been using Superslicer for my Voron because it has more of the features I want, although I've been trying to decide if it's worth switching to the 2.6 alpha because it also has a lot of really interesting features and I've been using it for my Prusa Mini and Ender 3 v2. I operate a print farm at work consisting of 6 Prusa i3s (one of which has an MMU) which we use Prusaslicer 2.5 for, a Markforged Mark Two which obviously uses Eiger, and some custom machines which we also normally use Prusaslicer for as well. I started with Cura and play with it occasionally, but only know it enough to help people with it a bit. As you'll see, most of the the features I would like are from Superslicer. 

The #1 thing I want and is keeping me from going to Prusaslicer is mouse ears. 
I absolutely love them and love Superslicer's implementation. I much prefer them over brims as I find that it's often just the corners that are needed and Superslicer's settings allow good tuning on where they end up. Less wasted time and material, but moreover easier to remove and can stay out of areas that are harder to clean up. The only flaw I've noticed in Superslicer's implementation (that I can remember at least) is that it would be nice if you could add more brim to corners instead of only having brim around corners. That way you would end up with a slightly dynamic and more efficient brim, with a smaller brim where it isn't as important and wider at corners and such. An alternative implementation I could see if you could be if you could change brims with modifier meshes such that it could be used to manually create mouse ears with a concentric pattern so it's easier to remove (could also just add the ability to change the solid infill pattern per part so you can add shapes to a similar effect). It would obviously be best to have both implementations, but that could end up being too much work and too many things to go wrong. 
Bonus points: I think it would be amazing if brim settings could be a material override so they can be automatically added for materials like ABS but removed for PLA so that users won't forget and lose a print. 

[For those who haven't seen Superslicer's mouse ears: in the brim settings, there is a checkbox to enable "mouse ears" which also enables you to change the max angle, detection radius, and pattern. When mouse ears are enabled, it acts as an override and will make a brim with the same settings as set with the normal options, but only around where it sees a change in angle measured within a given radius (not instantaneous change in angle but how much an angle changes within a given area).]

The main additional features that Superslicer has that Prusaslicer doesn't are sparse internal solid infill (additional thin supporting layer to help support top layers and reduce how much infill is needed for a consistent and solid surface) and of course some at least basic calibration tools. I'm sure calibration tools are a common request, but here are my two cents anyway. At a minimum, I would love to see Prusaslicer have first layer calibration (so it can be automatically sized to the bed), temperature calibration, retraction, and of course flow. I would love to see ironing, bridging, overhang, hole, and general dimensional accuracy calibration as well. It would be nice if max flow calibration was also built in, ideally CNC kitchen's tool in addition to a test pattern that doesn't require a milligram scale. It's a stretch, but it would also be nice if there were tools to easily tune things like encroachment, line spacing, and other advanced features, but that's probably a bit much. 

Other than that and things I imagine are already being worked on or have been talked about, the main things I would like are in the theme of adding control and ways to do things. I imagine most of these wouldn't be desired features since most are things that only the people who are really pushing their machines would use and so might just add bloat for most users, but nonetheless these are some of the main things I've noticed or come up with (gets more advanced and I imagine less likely to be done as the list goes on): 

- Solid infill settings 
I've had issues that looked like really bad pressure advance settings in that there was a lot of extra material sticking up where solid 
infill met perimeters and causing issues. that got fixed by changing the pattern to what Prusaslicer would probably call not 
connected rectilinear, so being able to change more than just the outermost layers would help with that. 
     
- Solid infill overlap/encroachment/anchor setting (separate from sparse infill)
Does what it says on the tin. I expect it could help reduce things like the hull line or other artifacts from solid infill affecting surface 
finish. It should also be useful for the issue I was facing above since that shouldn't exist for sparse infill (since extra material can 
move) so I'd rather ensure it's anchored properly and a separate consideration from solid infill. 
     
- Chamber temp settings
So it can be fed to macros more easily, or when not using macros to make it easier to park the hotend over the bed and use it as a 
chamber thermistor in the start gcode. 

- Layout customization
Very simple, would just be nice to be able to change the layout of things. Namely if settings in a category are in a single or multiple columns, but if one could change which tabs settings are under that could also be great. It would obviously be even more amazing if you could add more advanced and niche settings similar to cura.
     
- Speeds relative to max flow or other autopseed settings set in filament settings (or can otherwise be based on the material)
Would allow for things like fast or slow profiles to stay the same even if the material has different flow limits or characteristics. 

- Hardware/machine setting modifiers/multiples, especially for hotend and extruder related settings. 
It's more important if the above feature is implemented, but it would be great if you could do things like add an offset, multiplier, or override (or some combination of the three) based offset or whatever to things like temperature, retraction, extrusion multiplier, max flow rates, etc. based on the machine or nozzle. It could also be nice if there was a checkbox to enable an additional z offset for glue/adhesive coatings. This would make changing nozzles and such easier, but more importantly makes managing multiple machines (and multiple tools) much much easier since you could use the exact same material profiles for almost any hardware instead of them being hardware specific. Namely, adding a hotend temp offset for a steel vs brass nozzle, changing flow rate limits for different hotends and nozzles, adding some amount of extra/different retraction between different hotends/nozzles, etc. 
I imagine there would have to be some combination in both machine and material profiles, such as if the material profile will abide by those overrides but moreover if the overrides change based on the material. Perhaps there can be a set of overrides in the machine settings as well as conditions such as what kind of nozzle it has, direct or bowden, etc that the material profile can use to decide what overrides/modifiers pays attention to in what conditions, but also so it can use different overrides based on what hardware is present inste [... cut ...]  can be under advanced settings and not regularly seen or touched by most), but it it could make 3d printing damn close to plug and play. Lofty goal though. 

Sorry for the huge wall of text. I just wanted to get my ideas out there adequately enough for someone more skilled to at least consider them. I definitely don't expect many (or likely any) of this to be implemented, but let me know if there is anything I can do to clarify anything or correct mistakes. 

Thank you for coming to my ted talk

LiterallyKey, 2023-03-26:
I realized that I titled this as being a few small changes, but a lot are fairly big...

Bumblebiber, 2023-03-28:
> I'm definitely interested in "Dynamic flow compensation" as it would be easier than my current workaround – I use Gcode substitutions, for example if infill is printed at higher volumetric flow than perimeters:
> 
> > replace:
> > ;TYPE:Perimeter
> > with:
> > ;TYPE:Perimeter\nM221 S100 ; set flow to 100%
> 
> and
> 
> > replace:
> > ;TYPE:Solid infill
> > with:
> > ;TYPE:Solid infill\nM221 S110 ; set flow to 110%
> 
> etc.

This is the way! Also great for fan control (e.g. turn it on for outer perimeters only).

LiterallyKey, 2023-03-28:
> For improved mouse ears you can create yourself some "table tennis rackets" easily, save them as .3mf and add them to your model.

Yeah, but the problem is that they aren't concentric and instead are directly attatched and so harder to use without drawbacks. Having mouse ears as part of the brim generation allows it to be concentric and also have a seperatipn gap so you can use them with minimal artifacts on the finished part. Those also don't wrap around any outcroppings like Superslicer's do. Definitely better than nothing, but still not great IMO

Bumblebiber, 2023-03-28:
> Yeah, but the problem is that they aren't concentric and instead are directly attatched and so harder to use without drawbacks. Having mouse ears as part of the brim generation allows it to be [...]

You can add the concentric bottom infill feature as well

[image]

LiterallyKey, 2023-03-28:
> You can add the concentric bottom infill feature as well

Oh really? you can do that for just those sections and not the whole part? I guess I need to look closer and play with it more. I'm guessing saving them as 3mf instead of just adding a cylinder might help.

Bumblebiber, 2023-03-28:
> > You can add the concentric bottom infill feature as well
> 
> Oh really? you can do that for just those sections and not the whole part? I guess I need to look closer and play with it more. I'm guessing saving them as 3mf instead of just adding a cylinder might help.

Rightclick the object -> Add settings -> Infill 

Tick "Bottom fill pattern" (first list entry). Then you can change it to what ever you want on the right side.

Snuff1eupagus, 2024-04-26:
Clearly you have completely missed this ability, maybe you should watch this video.

https://www.youtube.com/watch?v=dhsbMmosbLU&t=4s

Enjoy

m10d, 2024-11-28:
There are multiple totally separate issues in this post.  OP should close and re-open individual issues, that's the way issue tracking software works.  This issue seems to have focused mostly on bunny ears which means it would be fine to edit the title & content to _just_ bunny ears, moving any small discussion of the other requests to new issues as necessary.

Tupson444, 2026-09-18:
still relevant