prusa3d/PrusaSlicer #10173: Crash when slicing with adaptive layer height

blastrock, 2023-03-27:
### Description of the bug

I want to use adaptive layer height on a model and PrusaSlicer crashes when I click on "Slice now".

### Project file & How to reproduce

I can't upload the file to github because "Something went really wrong". Here's a [google drive link](https://drive.google.com/file/d/1BJV0SjdpLHQV469fgnqGVWnaBP7WiqbK/view?usp=sharing).

1. Open the file
2. Select the model
3. Click on adaptive layer height in the top toolbar
4. Click on the "Adaptive" button in the bottom right dialog
5. Click on "slice now"
6. It crashes

Note that if I enable adaptive layer height, save the project, reopen it and slice it, it doesn't crash.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha5 (from flathub beta)

### Operating system

Linux Debian Unstable

### Printer model

SV06

kubispe1, 2023-03-29:
Fix of this issue will be part of the next release 2.6.0-alpha6
Thanks 
Closing the issue

closed 2023-03-29 by kubispe1

kubispe1, 2023-03-30:
@blastrock I forgot to ask you to retest it after release and let us know, if everything is fine. The issue was difficult to reproduce.