prusa3d/PrusaSlicer #10174: Prusa misrepresents Anycubic Predator as having reprap/sprinter instead of Marlin gcode

moosehaed, 2023-03-27:
### Description of the bug

Prusa misrepresents Anycubic Predator as having reprap/sprinter instead of Marlin gcode
Prusa slicer 2.6.0.alpha 5 
The profile for the Anycubic predator lists the firmware as reprep/sprinter.
The gcode should be Marlin.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 alpha 5 win64

### Operating system

windows 10

### Printer model

Anycubic Predator.

rtyr, 2023-03-28:
It should not be a problem and it could be even intentional. Do you have any issue related to that?

@nordurljosahvida FYI

moosehaed, 2023-03-28:
No issue as of now. I just added the anycubic predator and was going
through the settings and saw that gcode flavor. In other slicers is set to
marlin.

I used to lose my temper in arguments, until someone told me, 'Cooler heads
prevail.' Thus began my collection in the basement freezer."

rtyr, 2023-03-28:
OK I will close this as this flavor is pretty much identical to Marlin (legacy) with disabled machine limits. I also notified the author of the profiles, so he may re-evaluate it.

nordurljosahvida, 2023-06-19:
Hey all. Thanks @moosehaed for opening this and @rtyr for remembering and notifying me. good point. I too agree it should be corrected but there's no functional difference we can also keep it as it is. we completely redid our predators from scratch and now they all run on a duet board so unfortunately i won't be able to test any future profiles for stock predators, but i'm ofc available if my team and I can be of help. Thanks!