prusa3d/PrusaSlicer #10182: Internal infill stops short of perimeter (for some parts of the geometry)

AndKe, 2023-03-29:
### Description of the bug

internal infill stops short of the perimeter for some angles.

[image]

[top_protector.zip](https://github.com/prusa3d/PrusaSlicer/files/11100431/top_protector.zip)

### Project file & How to reproduce

slice the attached stl (inside zip file)
observe layer 11-13 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.1

### Operating system

Linux, Ubuntu 22.10

### Printer model

custom 500x500mm

neophyl, 2023-03-29:
Yes that's expected operation for PS with that STL.  It will put the other perimeters on top of the topfill.

Btw when supplying files for issues you need to normally attach a zipped up Prusa Slicer project file (3mf, using File>Save Project As).  That's usually because the settings are just as if not more important usually than the file you are trying to slice.

In this case though what you picture is what I would expect PS to generate from your model.

AndKe, 2023-03-29:
@neophyl - for project data: please see the attached top_protector_project.zip (in the post above)

I am referring to those gaps where certain areas do not meet the inner perimeter.  
I doubt you mean to tell me this is "expected" ?  ...on some parts of the part.

Marked it in green:

[image]

neophyl, 2023-03-29:
Sorry missed the first zip, only saw the one below the pic.

I've opened your project now and yes that's how PS slices that part and is expected to do so.  Its giving exactly what I would expect of the current software.  The current logic is that the infill continues between that in the lower into the upper sections.  The solid infill and topfill layers extend under the perimeters placed on top only so far.  

Its why you see several other issues opened on here where the topfill isn't unbroken layer under text on top of a plaque for example.  A quick search managed to find this one https://github.com/prusa3d/PrusaSlicer/issues/8818  which I recall from the time it was posted.

AndKe, 2023-03-29:
@neophyl Thank you. Why does the same layer in the "spokes" cover the whole area of the spoke, while the outer ring has this gap?

neophyl, 2023-03-29:
I think because the 'thickness' of those walls on the radial spokes is smaller than the walls on the rim.  The infill goes the same amount under the perimeters in both places but on those its enough to bring it all the way to the other perimeters.

closed 2026-09-29 by prusaslicer-bot[bot]