prusa3d/PrusaSlicer #10183: Access violation

jameskriebel, 2023-03-29:
### Description of the bug

While slicing a new object I received a "Access violation" for the first time.

[image]
[image]

### Project file & How to reproduce

This is a multicolor project, when slicing the error is produced at 90% sliced.
[D4-error.zip](https://github.com/prusa3d/PrusaSlicer/files/11101578/D4-error.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.1+win64

### Operating system

Windows 11

### Printer model

Ender 3 Pro with a 3D Chameleon

jameskriebel, 2023-03-29:
Possibly related to issue #10185, where I went back and re-sliced a previously successful project using the same config settings but the updated software version.  That too failed but with a different error.

kubispe1, 2023-03-29:
Hi, it is not only 2.5.1. There is an issue also in **2.6.0-alpha5+266**. On my Win 10, I have the same Access violation. We will check it. Thanks

lukasmatena, 2023-04-03:
@jameskriebel Thanks. The problem is caused by a mistake in your custom gcode:

`M109 R[ttemperature[next_extruder]} ; wait until temp is reached before resuming`
The combination of a legacy syntax using the square brackets, indexing something that is not a vector and the typo in the argument is too much for the placeholder parser to handle. The error check is missing in this case.

I just tried the same line in PrusaSlicer 2.2.0 and it crashes too. The bug must be there some many years. We will fix it.

jameskriebel, 2023-04-10:
Ok, can't expect your program to overcome my bad typing.  Found and fixed
the typo, now sliced just fine! Thanks for the help, and love the new error
in 2.5.2 which told me about my syntax error and where to find it.

Thanks!

closed 2023-04-14 by FidelCapo