prusa3d/PrusaSlicer #10185: Slicing error - possible permission issue?

jameskriebel, 2023-03-29:
### Description of the bug

Error encountered while slicing a multicolor project.  This may be related to the previous "Access violation" error I just reported.  As the new project failed I went back to a project that had sliced correctly in an older version of Prusa (I believe it was 2.4.1) and used all the same settings as the original slice.  This time I got a different error but still seems to be a permission problem.
[image]

### Project file & How to reproduce

Just attempting to slice the project as a multicolor object
[catan-error.zip](https://github.com/prusa3d/PrusaSlicer/files/11101797/catan-error.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.1+win64

### Operating system

Windows 11

### Printer model

Ender 3 Pro with a 3D Chameleon

jameskriebel, 2023-03-29:
Possibly related to issue #10183.

kubispe1, 2023-03-29:
:D Not for sure if it is related to 10183. This works for me in 2.5.1 and also in 2.6.0-alpha5+266. But as I mentioned, we will check it with the previous issue under a more detail view.

lukasmatena, 2023-04-03:
Definitely NOT related to #10183.

kubispe1, 2023-04-04:
@jameskriebel, do you have any program that checks changes on your drive? For example Google drive, One drive ... Or another sync app?
- Where is your file saved?

jameskriebel, 2023-04-10:
Moved the file to Local HDD, plus fixed the syntax error in 10183, it now
slices.

closed 2026-09-29 by prusaslicer-bot[bot]