prusa3d/PrusaSlicer #10186: Infill Failed error in alpha5 version

HolidayLighting, 2023-03-29:
### Description of the bug

[image]

This is with an STL from a paid Patreon, I'm not legally able to share it here but if required let me know and I will try and get permission

This happened with the alpha5 release with 15% Lighting Infill, Automatic painted on supports, Organic, 70% xy separation between an object and its support, and Avoid crossing curled overhangs.

### Project file & How to reproduce

This is with an STL from a paid Patreon, I'm not legally able to share it here but if required let me know and I will try and get permission

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha5+win64-202303061442_signed.zip

### Operating system

Windows 10

### Printer model

Prusa MK3s+

kubispe1, 2023-03-29:
Hi, please check the #10115 and #10055 could be a duplicate issues.

HolidayLighting, 2023-03-29:
> Hi, please check the #10115 and #10055 could be a duplicate issues.

Yes looks to be a possible duplicate of #10055 but it is not a duplicate of #10115  as no modifiers are being used.

Jan-Soustruznik, 2023-04-02:
Issue was closed as duplicate. Reported problem solved in a6

closed 2023-04-02 by Jan-Soustruznik