prusa3d/PrusaSlicer #10187: Precise layer height control

lukasradek, 2023-03-29:
**Is your feature request related to a problem? Please describe.**
I have an object with multiple "logical" vertical sections. Two would benefit from thin layers (one needs to be as translucent as possible the other is a single layer auxiliary support bridge) the rest from thick layers to cut print time.
`Variable layer height (VLH)` is not precise enough for that - neither in the "exact layer height" selection nor exact start and stop selection.
 
**Describe the solution you'd like**
It would be great to be able to specify exact values for layer height for exact height intervals.

Two solutions come to mind.
1. Add layer height value to modifiers (probably just the height range), even though I guess that would clash with VLH feature.
2. Extend the functionality of VLH.

**Describe how it would work**
Right now VLH is set up to be "smooth" and organic". It would be great to have more precise tools.
1. Add a vertical scale to the side of layer height chart. The scale would depict height (similar to slice preview).
3. Add a mouseover tooltip that would show the height user is pointing at (it could snap to the starting height of layers that are already visualized on current VLH graph). The tooltip is already in place, currently it shows layer height only.
Let's call the process of "Add detail", "Remove detail" etc. by clicking and dragging on the VLH chart `painting`.
4. Add option to paint precisely. Right now you can choose the size of painting area but the area is always feathered. Allow user to control the feather all the way to no feathering at all. The best would be to let user input `height of the brush` and `feather percentage`. Height of the brush could be specified as `layer count` or `height`. If it was specified as layer count, the brush would snap to layer visualization.
5. Add option to paint precise layer height values. Allow user to input a precise layer height that would be painted on the chart.

This would allow user to have much more precise control over layer height and would make the process of setting large vertical areas to precise value much easier (possible).

**Describe alternatives you've considered**
Mentioned above.

neophyl, 2023-03-29:
Just use several height range modifiers and specify the print height in each section.  The printing height is already part of a height range modifier.  Precise control of the start and stop heights and should do what is needed.

lukasradek, 2023-03-29:
Jesus... my mind just unblocked 😁.
I was staring and wondering about the `Layer height` in `Settings for height range` for a long time and since it is quite out of place (regarding modifier properties) it never occured to me that it changes the layer height .😁

Thanks.

Anyway, the variable layer height expansion would be cool still 🙂.

closed 2026-09-02 by lukasradek